rusefi/rusefi #6022: who makes what

rusefillc, 2024-02-22:
"make clean" in firmware cleans simulator now.

Let's spell out what's the plan

What about building - what command from what folder would make simulator?
What about building - what command from what folder would make unit_tests?

i.e. things used to be pretty decoupled between three folders. How do I compile only firmware if I do not care to compile simulator?

```
rusefi@DESKTOP-57S88S8:~/rusefi/firmware$ make clean
PROJECT_BOARD: f407-discovery
BOARD_DIR:     ././config/boards/f407-discovery
PROJECT_CPU:   ARCH_STM32F4
SHORT_BOARD_NAME: f407-discovery
CPU_HWLAYER:   ports/stm32/stm32f4
CONFDIR:       ./hw_layer/ports/stm32/stm32f4/cfg
LDSCRIPT:      ./hw_layer/ports/stm32/stm32f4/STM32F4.ld
bundle.mk:132: warning: overriding recipe for target '&'
bundle.mk:112: warning: ignoring old recipe for target '&'
Cleaning PCH
rm -f ./pch/pch.h.gch/firmware
Cleaning Bundle
make -C ../simulator clean
make[1]: Entering directory '/home/rusefi/rusefi/simulator'
OS is []
Cleaning Simulator
rm -rf build
rm -rf .dep
Done Cleaning Simulator
Cleaning
- .dep
- build

Done
make[1]: Leaving directory '/home/rusefi/rusefi/simulator'
BOARD_DIR=../././config/boards/f407-discovery BOARD_META_PATH=../ make -C bootloader clean
make[1]: Entering directory '/home/rusefi/rusefi/firmware/bootloader'
Cleaning PCH
rm -f ../pch/pch.h.gch/bootloader
Cleaning-BLT
rm -fR blbuild/*
Cleaning-GCC
- .dep
- blbuild

Done
make[1]: Leaving directory '/home/rusefi/rusefi/firmware/bootloader'
rm -rf rusefi.snapshot.f407-discovery
Done Cleaning Bundle
Cleaning-GCC
- .dep
- build

Done
rusefi@DESKTOP-57S88S8:~/rusefi/firmware$
```

chuckwagoncomputing, 2024-02-22:
`make` makes firmware
`make bundle` makes everything (but not yet)

rusefillc, 2024-02-22:
> Would it be more expected to have separate `clean` and `clean-bundle` commands?

maybe one day but let's live as is. My main focus would be to make sure that building is more separate.

chuckwagoncomputing, 2024-02-22:
The plan is to have `make` generate configs if they are needed. For example, if you modify some .c file, the configs won't generate when you run make, but if you modify rusefi_config.txt, they will.

rusefillc, 2024-02-22:
> The plan is to have `make` generate configs if they are needed. For example, if you modify some .c file, the configs won't generate when you run make, but if you modify rusefi_config.txt, they will.

yes that's very much desired since well, YES.

i mean do not build simulator while building firmware or unit_tests

chuckwagoncomputing, 2024-02-22:
OK we are on the same page. 👍

chuckwagoncomputing, 2024-03-09:
Fixed with:
https://github.com/rusefi/rusefi/commit/20972847de3821ee38ceb441f0ae76714ad82a7d
https://github.com/rusefi/rusefi_documentation/pull/504

closed 2024-03-09 by chuckwagoncomputing