rusefi/rusefi #6032: restore POST_BUILD_SCRIPT

rusefillc, 2024-02-23:
https://github.com/rusefi/rusefi/pull/5869/files feature seems to be lost

it's desired to have an option of a script hook between binary compilation and bundle packaging

rusefillc, 2024-02-23:
@chuckwagoncomputing would it be somewhere around

```
.h2d-sentinel: $(BUILDDIR)/$(PROJECT).hex $(BOOTLOADER_OUT) $(BINSRC) | $(DELIVER)
ifeq ($(USE_OPENBLT),yes)
	$(H2D) -i $(BOOTLOADER_HEX) -i $(BUILDDIR)/$(PROJECT).hex -C 0x1C -o $(DFU) -b $(DBIN)
else
	$(H2D) -i $(BUILDDIR)/$(PROJECT).hex -C 0x1C -o $(DFU)
	cp $(BUILDDIR)/$(PROJECT).bin $(DBIN)
endif
	@touch $@
```

i.e. is that more or less where we do binary post-compilation processing? Can we add a configurable hook around here?

chuckwagoncomputing, 2024-02-23:
xyproblem.info

chuckwagoncomputing, 2024-02-24:
I'm thinking the best place to call a post-build script would be right before zipping the bundles.

https://github.com/rusefi/rusefi/blob/8f60601a3d103c047c4ad1691ea97c9aff051728/firmware/bundle.mk#L164-L168

andreika-git, 2024-02-25:
I think it's a good place for the post-build script!
And can you please also add a pre-build script (just before calling make) the same way as the post-build one?

chuckwagoncomputing, 2024-03-01:
I still don't know what the goal of a pre-build script would be, but as of now I don't think the makefile is the right place for it.

rusefillc, 2024-03-01:
POST_BUILD_SCRIPT was restored

if we need anything else we would need a separate clear ticket

closed 2024-03-01 by rusefillc