rusefi/rusefi #6040: 'make' build of unit tests is missing something

rusefillc, 2024-02-24:
just added new bit field https://github.com/rusefi/rusefi/commit/8f60601a3d103c047c4ad1691ea97c9aff051728

github master is green

git pull
cd unit_tests
make

```
rusefi@DESKTOP-57S88S8:~/rusefi/unit_tests$ git pull
Already up to date.
rusefi@DESKTOP-57S88S8:~/rusefi/unit_tests$ make -j12
Compiling value_lookup_generated.cpp
../firmware/controllers/lua/generated/value_lookup_generated.cpp: In function ‘float getConfigValueByName(const char*)’:
../firmware/controllers/lua/generated/value_lookup_generated.cpp:1089:32: error: ‘struct engine_configuration_s’ has no member named ‘luaCanRxWorkaround’
 1089 |    return engineConfiguration->luaCanRxWorkaround;
      |                                ^~~~~~~~~~~~~~~~~~
../firmware/controllers/lua/generated/value_lookup_generated.cpp: In function ‘bool setConfigValueByName(const char*, float)’:
../firmware/controllers/lua/generated/value_lookup_generated.cpp:3585:24: error: ‘struct engine_configuration_s’ has no member named ‘luaCanRxWorkaround’
 3585 |   engineConfiguration->luaCanRxWorkaround = (int)value;
      |                        ^~~~~~~~~~~~~~~~~~
At global scope:
cc1plus: warning: unrecognized command line option ‘-Wno-unused-private-field’
make: *** [../firmware/../unit_tests/rules.mk:113: build/obj/value_lookup_generated.o] Error 1
```

chuckwagoncomputing, 2024-02-24:
`rm`ing pch/pch.h.gch/unit_tests helps.
It seems that make doesn't know that it needs to re-build pch.h.

chuckwagoncomputing, 2024-02-24:
I finally figured out how to get back to a broken state.
```
git checkout 14c0e4d
make -j$(nproc)
rm .dep/*
git checkout db4280d
make -j$(nproc)
```

chuckwagoncomputing, 2024-02-24:
This can happen when building firmware too.
Here's a failing firmware build run for core8: https://github.com/chuckwagoncomputing/fw-custom-core8/actions/runs/7946920817
This was on rusEFI commit c4e4165

chuckwagoncomputing, 2024-02-25:
Here are some notes from past self solving a similar problem in the firmware build for a future PR:

> There's one thing I don't really like here.
> ```
> $(OBJS): $(CONFIG_FILES)
> $(PCHOBJ): $(CONFIG_FILES)
> ```
> This is kinda following the example of rusefi_pch.mk
> https://github.com/rusefi/rusefi/blob/44ebfc633f5c5d29b583b38a130f25ce64031cd6/firmware/rusefi_pch.mk#L20-L21
> 
> This is required because of the way the ChibiOS makefile is set up. The automatic prerequisite generation is done in the same step as the compilation. Therefore, if the .dep folder isn't already in the correct state, Make doesn't know which .o files require the config headers until it has already tried and failed to compile them. So to work around this, we just tell Make that all .o files require the config headers. The downside of this is that if a config header changes, all .o files will be rebuilt, not jut those that require the config headers.

Back to current self:
I propose I move all config-generation to its own makefile, which the unit test makefile can include, and make CONFIG_FILES a prerequisite of PCHOBJ in rusefi_pch.mk.
What I don't understand is what changed to cause this issue.
@rusefillc Have you habitually run make in unit_tests without running make clean first?

rusefillc, 2024-02-25:
@chuckwagoncomputing i hit clean in case of any doubt. For this issue clean did not seem to help?

chuckwagoncomputing, 2024-02-25:
OH I think the answer is that I broke `make clean`

chuckwagoncomputing, 2024-02-25:
But I still want to fix the actual issue in order to not require `make clean` every time.

chuckwagoncomputing, 2024-02-25:
https://github.com/rusefi/rusefi/pull/6042

rusefillc, 2024-02-29:
@chuckwagoncomputing do we still need this one?

chuckwagoncomputing, 2024-03-01:
This issue is solved.

closed 2024-03-01 by chuckwagoncomputing