rusefi/rusefi #6041: better than luaCanRxWorkaround solution for lua CAN RX performance problem

rusefillc, 2024-02-24:
problem: https://github.com/rusefi/rusefi/issues/5841 due to extreme cost of ``lua_newtable`` on the order of 500us/0.5ms see https://github.com/rusefi/rusefi/issues/5841#issuecomment-1961980005

temporary quick solution https://github.com/rusefi/rusefi/commit/f6ca672852af2e44e3b84f79eafaf63acdbcb2c2

a more elegant solution is desired:
* automatically define global_can_data
* maybe: define global_can_data only if CAN RX functionality is used to reduce Lua space usage?
* prevent GC of that variable maybe see https://stackoverflow.com/questions/66807244/lua-c-is-there-a-way-to-keep-a-value-in-c-but-still-have-it-not-garbage-colle

mck1117, 2024-02-26:
you may be interested in https://github.com/FOME-Tech/fome-fw/commit/d10e274153e2bd35088946f3286bce995f8b71c5

mck1117, 2024-02-26:
last column is average ms for the table create + writing values

[image]

rusefillc, 2024-02-26:
> last column is average ms for the table create + writing values

yes and no - skipping GC is cool but garbage is still being populated and GC is in createtable by default probably because tables are heavy? You measurements are missing total GC times, are those available to us?

mck1117, 2024-02-26:
It doesn't skip the gc entirely. It skips calling it in the tight loop of doing CAN reception, deferring it to happen later (which it does).

"total ms" is `doLuaCanRx` and `invokeTick`, doing 25 CAN frames every time with empty callbacks on the Lua side.

dron0gus, 2025-11-09:
Should we pass CAN messages to/from Lua as userdata?

See https://www.lua.org/pil/28.1.html