rusefi/rusefi #6043: openblt via CAN

rusefillc, 2024-02-25:
So many openblt changes recently can we get small-can-board firmware upload to work?

andreika-git, 2024-03-03:
I've tested the possibility to flash firmware via ST Bootloader on CAN2 using STLinkv3 as it is useful to be able to flash blank chips from the factory via CAN.

TLDR; Interim results are disappointing. So far I haven't been able to get it to work...

Setup:
- 2chan rev.F (mega-mcu144)
[image]

- STLink-V3MODS
 [image]
   * The pinout is:
  Pin 8: GND
  Pin 9: Bridge CAN RX (Input)
  Pin 10: Bridge CAN TX (Output)
  Pin 22: 5V (used for CAN transceiver module power supply)
   * See https://www.st.com/resource/en/user_manual/um2502-stlinkv3mods-and-stlinkv3mini-debuggerprogrammer-tiny-probes-for-stm32-microcontrollers-stmicroelectronics.pdf )

- CAN transceiver module based on MCP2551
[image]

- STM32CubeProgrammer v2.15.0 (Win64)

Since the bootloader uses only specific pins PB5 (Rx) and PB13 (Tx) for CAN2, and on 2chan we use pins PB12 (SPI2_CS) PB13 (SPI2_SCK), I had to modify the 2chan board by cutting the CAN-RX trace and connecting it with wire to SPI1_MOSI (PB5). The CAN-Tx was already connected correctly. Also, the transceiver in the CAN2 module was not soldered on the board - I had to solder it.

[image]
[image]
[image]

The settings for the STM32CubeProgrammer were used by default, with the exception of the bitrate, which according to AN3154 should be 125kbps.

[image]

The data is sent via the CAN bus to the transceiver and to the STM32 (checked with an oscilloscope on pin PB5), and the MCU does not produce anything in response.

[image]

I have a hypothesis that this may be related to the quartz frequency. The AN3154 has the following operating algorithm for the CAN bootloader (Figure 2):

[image]

It can be seen that they proceed from the assumption that the frequency of the external quartz is either 8 MHz, or 25 MHz, or 14.7456 MHz, and set the bitrate based on this. We have 20 MHz quartz in our mcu module... I can’t check this yet, because I don’t have quartz in the right package for these frequencies...

andreika-git, 2024-03-08:
It works!

Connected to CAN2 (PB5+PB13) the same way as for the ST bootloader (see #5882):
[image]

It requires some modifications to our openblt code (PR is getting ready). Also we need to somehow switch to "OpenBLT mode" (I used a USB connection and corresponding Console menu option for the switch).
The command line is as following:
```
BootCommander.exe -t=xcp_can -d=peak_pcanusb -c=0 -b=500000 -tid=667h -rid=7E1h rusefi_update.srec
```
The output is:
[image]

rusefillc, 2024-03-10:
TODO
1. get working on proteus - PR into master if needed
2. get working on small can board - PR into master if needed. I assume there is little change of flash shortage here?

andreika-git, 2024-03-11:
@rusefillc On small can board, do we need OpenBLT on CAN1 or CAN2?

rusefillc, 2024-03-11:
@andreika-git can1? But really whichever we can get working with least effort

andreika-git, 2024-03-12:
OpenBLT+CAN works on small-can-board (CAN1)!
We'll need to re-test it when the next official fw bundle will be ready for the small-can-board.

andreika-git, 2024-03-12:
Confirmed IT WORKS! with official GHA bundle on small-can-board (https://github.com/rusefi/rusefi/actions/runs/8250061030/artifacts/1318759510)

rusefillc, 2024-03-12:
Worked in NJ using master!

* manually switch to BLT mode (which needs nicer red blinking on small-can-board @andreika-git )

[image]

closed 2024-03-12 by rusefillc