rusefi/rusefi #6044: first round of documentation

rusefillc, 2024-02-25:
A lot of great progress with the Makefile recently

how do I know what targets are available? 

https://stackoverflow.com/questions/4219255/how-do-you-get-the-list-of-targets-in-a-makefile mentions make 4.4 which is not a package for any distro yet? sounds like we need some sort of help?

specific issue which triggered the ask - I have small-can-board failing to compile with blt, maybe blt itself is failing? I would love to make clean and make only bootloader.

rusefillc, 2024-02-25:
@chuckwagoncomputing an example of a more specific question https://github.com/rusefi/rusefi/commit/194d35c7d42dfec855ad8a00d6431dabd5be77e5

chuckwagoncomputing, 2024-03-09:
Fixed with:
https://github.com/rusefi/rusefi/commit/20972847de3821ee38ceb441f0ae76714ad82a7d

closed 2024-03-09 by chuckwagoncomputing