rusefi/rusefi #6067: Windows file access issue somewhere

rusefillc, 2024-02-27:
@chuckwagoncomputing how can I see https://github.com/chuckwagoncomputing/rusefi/actions/runs/8059487251/job/22013988308 on master? i.e. is that not a problem on master yet?

rusefillc, 2024-02-27:
@chuckwagoncomputing need your help https://github.com/rusefi/rusefi/commit/899f7f643913a966caac05bfee8b945b27ba3a26

chuckwagoncomputing, 2024-02-27:
Re: `cmp ${TEMP_FILE} ${SIGNATURE_FILE_NAME} || mv -f ${TEMP_FILE} ${SIGNATURE_FILE_NAME}`
cmp compares files to see if they are different. If `SIGNATURE_FILE_NAME` is missing it will print an error but, the `mv` will still happen as it should. I thought about adding `2>/dev/null` to the cmp command to silence the error, but I haven't done it yet.
The long way of writing this would be:
```
if ! cmp ${TEMP_FILE} ${SIGNATURE_FILE_NAME}; then
  mv -f ${TEMP_FILE} ${SIGNATURE_FILE_NAME}
fi
```

That run is on my bootloader-configs branch, not master. It's not a problem on master because we're not trying to use full paths. Java is probably turning `config/boards/foo/bar` into `config\boards\foo\bar`, but cygwin is fine with relative paths using `\`, just not absolute paths for some reason.
Testing on my computer at work:
```
AzureAD+DavidHoldeman@MAINTENANCE-MIN MSYS /c/Users
$ ls 'DavidHoldeman/test'
hello  world

AzureAD+DavidHoldeman@MAINTENANCE-MIN MSYS /c/Users
$ ls 'DavidHoldeman\test'
hello  world

AzureAD+DavidHoldeman@MAINTENANCE-MIN MSYS /c/Users
$ ls '/c/Users/DavidHoldeman/test'
hello  world

AzureAD+DavidHoldeman@MAINTENANCE-MIN MSYS /c/Users
$ ls '\c\Users\DavidHoldeman\test'
ls: cannot access '\c\Users\DavidHoldeman\test': No such file or directory

AzureAD+DavidHoldeman@MAINTENANCE-MIN MSYS /c/Users
$ ls '/c\Users\DavidHoldeman\test'
hello  world

```