rusefi/rusefi #6069: Document and/or fix LTS upload paths

rusefillc, 2024-02-27:
I hit "Run Workflow with LTS master" and I see files appearing at folders like https://rusefi.com/build_server/lts/alphax-8chan/ with "master" anywhere on path

I see older (?) https://rusefi.com/build_server/lts/lts-2023.01/ and https://rusefi.com/build_server/lts/master/ folders which seem to use different approach

Specifically right now I am looking to make a release candidate. To make a release candidate I need a set of freshly fully build bundles ideally all in order folder not subfolders.

https://github.com/rusefi/rusefi/wiki/Dev-LTS-project seems to be the page to update?

[image]

rusefillc, 2024-02-27:
@chuckwagoncomputing can you help?

chuckwagoncomputing, 2024-02-27:
I broke this!

PR #6071 (2024-02-27): Fix LTS upload folder
fix #6069
closed 2024-02-27 by rusefillc