rusefi/rusefi #6079: UAEFI crash if set to daihatsu 4cyl trigger

runesmdk, 2024-02-28:
Board crashes and looses comms (fail to burn page) if set to daihatsu 4cyl trigger using c5 cam1/hall1 input.

runesmdk, 2024-02-28:
> Yes, I am able to reproduce the issue, thank you for the report!
> 
> @runesmdk please use https://github.com/rusefi/rusefi/releases/tag/2024-02-18 as a workaround that one seems to work

I downloaded the fresh snapshot today. This link is to nightly's thats 2 weeks old?

rusefillc, 2024-02-28:
> This link is to nightly's thats 2 weeks old?

yes

https://github.com/rusefi/rusefi/releases/tag/2024-02-22 does not have burn issue but has 5VP issue

actually https://github.com/rusefi/rusefi/releases/tag/2024-02-27 seems to be OK as well so whatever defect we have introduced seems really recent?

runesmdk, 2024-02-28:
> > This link is to nightly's thats 2 weeks old?
> 
> yes
> 
> https://github.com/rusefi/rusefi/releases/tag/2024-02-22 does not have burn issue but has 5VP issue
> 
> actually https://github.com/rusefi/rusefi/releases/tag/2024-02-27 seems to be OK as well so whatever defect we have introduced seems really recent?

Will try later, thanks

runesmdk, 2024-02-29:
> Will try later, thanks

Updated to 2024-02-27. Kept getting "Data write failed for 2 attempts, local data will be refreshed from controller" and "Controller error: Burn page failed crc check" while using "slow comms (wireless)" on usb cable. Switched to "fast comms" and worked. Dont know if i its just because you cant use slow on usb? But daihatsu trigger is now set atleast.

rusefillc, 2024-03-01:
Does not reproduce with f407 discovery or proteus F7 firmware.

I can reproduce ONLY with uaefi firmware :(

On any TS burn a disconnect happens but uaEFI keeps blinking blue LED. Even rusEFI console is no longer to connect to unit until power cycle. 

So, happy rusEFI console. TS burn breaks comms. Unhappy rusEFI console.

rusefillc, 2024-03-01:
Does NOT happen with uaEFI firmware on stm32f407 discovery hardware

rusefillc, 2024-03-01:
DOES happen on uaEFI even after full chip erase.

closed 2024-03-02 by rusefillc