rusefi/rusefi #6081: custom repo Unable to access config_definition-all.jar while still downloading gradle

rusefillc, 2024-02-28:
```
Error: Unable to access jarfile ../java_tools/configuration_definition/build/libs/config_definition-all.jar
make: *** [rusefi_config.mk:29: .config-sentinel] Error 1
make: *** Waiting for unfinished jobs....
Downloading https://services.gradle.org/distributions/gradle-8.6-all.zip
....................10%.....................20%.....................30%.....................40%.....................50%.....................60%.....................70%.....................80%.....................90%.....................100%

Welcome to Gradle 8.6!
```

rusefillc, 2024-02-28:
@chuckwagoncomputing I have a human confirming same while trying to build this rusefi on a Mac

[image]

rusefillc, 2024-02-28:
@chuckwagoncomputing I got this locally on Windows

[image]

chuckwagoncomputing, 2024-02-28:
Fix is coming up

PR #6087 (2024-02-28): Regenerate configs if any config files have changed
This should fix #6081 and #6082.
closed 2024-02-28 by rusefillc