rusefi/rusefi #6100: RFC: Unified list of build targets

chuckwagoncomputing, 2024-02-29:
Currently we have two lists of build targets: the matrix in https://github.com/rusefi/rusefi/blob/master/.github/workflows/build-firmware.yaml and the list in https://github.com/rusefi/rusefi/blob/master/firmware/gen_config.sh.

These lists have different keys. The matrix has board directory, bunde name/build target, and skip rate. The gen_config list has board directory and short board name.

I propose we add a list file with the following keys: meta-info file and skip rate.

It is possible to generate the build matrix for the Firmware at GHA workflow on the fly, pulling the information from a file in the repo. The board directory is always the directory the meta-info file is in. The bundle name could be taken from the meta-info file name, or assumed to be the same as SHORT_BOARD_NAME if the meta-info file is the default meta-info.env. I haven't rigorously checked, so I'm not sure if there are any counterexamples. If the skip rate is 100, it is not even added to the matrix. The skip rate processing could even be done in the matrix generation step, so that only those boards that were not skipped would be visible in the sidebar, so that you wouldn't be misled by thinking a build had succeeded when in fact it had been skipped.

gen_config would have to load the meta-info files to get the SHORT_BOARD_NAME. This would add a small amount of overhead, but I don't think it's worth counting.

More ideas:
The skip rate could be moved into the meta-info file, leaving a list of meta-info files.
If this is done, the list could be done with altogether, instead just searching for meta-info*.env files within config/boards.
The downside of this is that there could be a case where you were adding a board but didn't want anything done with it yet. You could name the meta-info file something else temporarily.

chuckwagoncomputing, 2024-02-29:
Also, small fun fact: The short board name for config/boards/nucleo_h743 in gen_config.sh is "nucleo_h743", but in config/boards/nucleo_h743/meta-info.env it's "stm32h743_nucleo", as I copied it [from build_firmware.yaml.](https://github.com/chuckwagoncomputing/rusefi/commit/4ae60f4d36ac8042ec909effa2e664ab07f2c223#diff-fd6a34015e12e374d49724f684d3b6cc6088f0a4dd911afcde45c2c93d24ff04L395)

rusefillc, 2024-02-29:
See also https://github.com/rusefi/rusefi/issues/5546

rusefillc, 2024-02-29:
> nucleo

those are officially a mess, I know I did _some_ incompatible change a couple of weeks ago. Just use any common sense, no value in protecting any specific names for nucleo

rusefillc, 2024-02-29:
> I propose we add a list file with the following keys: meta-info file and skip rate.

short answer is: yes, please! https://github.com/rusefi/rusefi/issues/5546

apparently we might actually have a totally out of date dead [boards.yaml](https://github.com/rusefi/rusefi/commit/fca6319803910a925486f31a69e2ab094874589b#diff-1c9c498610421fa3bbe725269a0fe7fa4e16f332cca7cd804c764a11f20913f7)?! I recall I got distracted after reading a bit about yaml to bash importing via regex and no official regex seems to exist?

chuckwagoncomputing, 2024-02-29:
I refuse to try to parse yaml with regex. Using a tool like https://github.com/mikefarah/yq as I do with https://github.com/chuckwagoncomputing/interactive-pinout is the only acceptable option.
However, if we only have one or two keys, CSV or TSV is a good option, as it's much easier to parse with bash. It's basically what gen_config.sh is already doing.

chuckwagoncomputing, 2024-03-08:
For gen_config.sh, do we want to generate config for all boards that do not have SKIP_RATE=100, or are there exceptions to that?

rusefillc, 2024-03-08:
@chuckwagoncomputing I've say generate config for all regardless of SKIP_RATE

SKIP_RATE saves real GHA resources. gen_config is lightning fast comparing with full board build.

PR #6178 (2024-03-09): Automatically find meta info files for gen config
Fix #6100
Run: https://github.com/chuckwagoncomputing/rusefi/actions/runs/8210121705
Commit: https://github.com/chuckwagoncomputing/rusefi/commit/a6bd10c2c0ba1b15aa2d88b4571c26f5367b1344
closed 2024-03-09 by rusefillc