rusefi/rusefi #6103: multi-runner multi-target HW CI

rusefillc, 2024-03-01:
In order to run more than one build-target on a device we need to filter both st-link and ECU by serial number

At the moment we have those serials hard-coded .cfg and .yaml file perspectively

In order to run more than one runner per target while having more than one build-target per running we would need to provide each build-target with a list of allowed st-link and ECU serials so that runner locates the local white listed device

rusefillc, 2024-03-01:
https://github.com/a-v-s/gethla is pretty cool

chuckwagoncomputing, 2024-03-09:
I closed #6141 because I don't think it's the right way forward.
I propose we use environment variables, which could be passed to a docker runner.
```
HARDWARE_CI_SERIAL_DEVICE
HARDWARE_CI_STLINK_SERIAL
HARDWARE_CI_VBATT
```

chuckwagoncomputing, 2024-03-09:
Apparently the GHA runner doesn't inherit the environment, you have to put the variables in .env file.
https://stackoverflow.com/questions/71179099/self-hosted-environment-variables-not-available-to-github-actions

closed 2024-03-13 by rusefillc