rusefi/rusefi #6114: a way to build older custom firmware, I guess a step towards custom LTS

rusefillc, 2024-03-02:
* say master is broken
* say one day in the future I want to build https://github.com/rusefi/fw-custom-paralela/ from the past

here's the problem:
https://github.com/rusefi/fw-custom-paralela/blob/master/.github/workflows/build-firmware.yaml points at rusefi/rusefi/.github/workflows/custom-board-build.yaml@ *master*

rusefillc, 2024-03-02:
@chuckwagoncomputing is this something you can help with?

chuckwagoncomputing, 2024-03-03:
I thought this could be done by having a job that runs first that gets the ref of the submodule.
However, this doesn't work.
```
Invalid workflow file: .github/workflows/build-firmware.yaml#L30
error parsing called workflow
".github/workflows/build-firmware.yaml"
-> "rusefi/rusefi/.github/workflows/custom-board-build.yaml@${{needs.set-ref.outputs.ref}}"
: failed to fetch workflow: reference to workflow should be either a valid branch, tag, or commit
```

Option B is to turn custom_board_build.yaml into an action rather than a workflow, then checkout the submodule in build_firmware.yaml and call the action locally, i.e. `uses: ./ext/rusefi/.github/workflows/custom_board_build.yaml`.
I think this will work but I haven't tried it. It will require updating every custom board repo's build_firmware.yaml as it is a breaking change.

rusefillc, 2024-03-03:
@chuckwagoncomputing breaking change is zero problem since very few custom repos

chuckwagoncomputing, 2024-03-05:
Test run: https://github.com/chuckwagoncomputing/fw-custom-example/actions/runs/8157031584

PR #6162 (2024-03-08): convert custom-board-build to an action
Fix #6114

Downsides:
- In the Github interface, the steps don't have an individual collapsible section, they all share one section.
- The path to the rusefi submodule is hard-coded, so if someone changes it they will have to change it there as well as passing rusefi_dir to the action.
https://github.com/chuckwagoncomputing/fw-custom-example/blob/1a8d55ec981fe55b66de9e13746e560ae9e8173e/.github/workflows/build-firmware.yaml#L25
- Actions don't have specific inputs for secrets, so secrets are passed as normal inputs then masked with the `::add-mask` GHA c [... cut]
closed 2024-03-08 by rusefillc