rusefi/rusefi #6116: UAEFI Burn page 0 failed from full flash erase and changing reasonable settings

ggurov, 2024-03-02:
**Describe the bug**
From a full chip erase and fresh firmware flash, unsetting intake/exhaust cam sensor pins causes a TS error

**Hardware**
 - Board: UAEFI
 - Bench

**To Reproduce**
Steps to reproduce the behavior:
1. Fully erase UAEFI
2. Flash latest firmware (2024-03-02)
3. Open trigger setup menu
4. change cam sensor bank 1 intake/cam sensor bank 1 exhaust to "NONE"
5. click Burn
6. See error

**Expected behavior**
No error is produced, and settings are updated.

**Screenshots**

[image]

[image]

[image]

**Additional context**
Controller keeps flashing BLUE

closing TS shows:

[image]

restarting TS - controller cannot reconnect

after controller reset, TS connects, and settings are not updated:

[image]

rusefillc, 2024-03-02:
Great report! Looks like a dup of #6079

rusefillc, 2024-03-02:
@ggurov please test with revent commit

closed 2024-03-02 by rusefillc