rusefi/rusefi #6147: 2ZZ 36-2 with  custom hardware VR giving sync issues, Logic log inside!

ElDominio, 2024-03-05:
**Describe the bug**
The car has issues at about 2800 RPM, yet logic data seems fine? Shows instant RPM as 0 though on log

**Hardware**
 - Board: Paralela with this https://rusefi.com/forum/viewtopic.php?t=1115 hardware to decode  VR
 - Engine info Toyota 2zzFE
 
**To Reproduce**
Replay logic log?

**Expected behavior**
We keep reading RPM without fault

**Additional context**
I have tested this hardware in a Honda B series to 7500rpm and it reads without issue (it is a 24+1 on cam, so technically 12+1). I did a logic  analyzer and a normal TS datalog, they are attached. The Logic log looks fine, need more opinions

[DataLogs.zip](https://github.com/rusefi/rusefi/files/14502092/DataLogs.zip)

rusefillc, 2024-03-07:
@ElDominio 100Hz logs bad for trigger investigation, max data rate logs good for trigger investigation see https://github.com/rusefi/rusefi/wiki/HOWTO-upload-log#data-rate

rusefillc, 2024-03-07:
@ElDominio logs clearly show unexpected sync latest ratio0 you can raise gap range using override

Until this is reproduced on official hardware I consider this a hardware problem, not sure what action are you looking for?

[image]

closed 2024-03-07 by rusefillc

ElDominio, 2024-03-07:
[image]

Not having my logs at max data rate is a very old problem, i always use max data rate since I was reminded of that every single time.

This is not reproducible on official hardware because official hardware does not work with shared ground VR sensors which is what Toyota/Honda use up to very recently.