rusefi/rusefi #6175: Can we LTS?

rusefillc, 2024-03-08:
Right now master does not even update artifacts

At the same time we have https://github.com/rusefi/rusefi/releases/tag/release_20240304 needs https://github.com/rusefi/rusefi/commit/09c7619dc7007d4e21ba49de7781201f55e4cfaa and https://github.com/rusefi/rusefi/commit/09c7619dc7007d4e21ba49de7781201f55e4cfaa

As is I am not sure what the procedure I know we have https://github.com/rusefi/rusefi/wiki/Dev-LTS-project and https://github.com/rusefi/rusefi/wiki/Internal-Technical-Document-Long-Term-Support-LTS

oh actually https://github.com/rusefi/rusefi/wiki/Internal-Technical-Document-Long-Term-Support-LTS was really helpful I know what's the first step!

rusefillc, 2024-03-08:
https://github.com/rusefi/rusefi/tree/lts-2024.03a created

rusefillc, 2024-03-08:
renaming https://github.com/rusefi/rusefi/tree/lts-202403a and updating documentation

rusefillc, 2024-03-08:
@chuckwagoncomputing looks like yes we can!

Documentation was improved, bundles have proper folder name inside .zip for instance ``rusefi.lts-202403a.mre_f4``

chuckwagoncomputing, 2024-03-09:
Nice!

closed 2024-03-09 by chuckwagoncomputing