rusefi/rusefi #6195: custom board canned tune process would need board-specific x86 binary (simulator or unit_tests)

rusefillc, 2024-03-10:
See https://github.com/rusefi/rusefi/wiki/Canned-Tune-Process#internal-implementation-details

Same as this main repo exports current engine-specific tune, custom boards should be exporting board-specific tunes.

rusefillc, 2024-03-10:
For example fw-custom-paralela now has custom default tune https://github.com/rusefi/fw-custom-paralela/commit/f7b99d2b5194e3d1554823de08a19caffcd6cd11 which uses https://github.com/rusefi/rusefi/commit/fbdad77e58e8be69341c45c56bd16ecbecae5f28 hook

I assume next step would be to compile simulator from within fw-custom-paralela with https://github.com/rusefi/fw-custom-paralela/blob/master/default_tune.cpp added into build

Executing such a simulator with ``./build/rusefi_simulator 10`` similar to https://github.com/rusefi/rusefi/blob/master/.github/workflows/build-simulator.yaml would produce default tune with https://github.com/rusefi/fw-custom-paralela/blob/master/default_tune.cpp accounted for, the rest I can deal with easily :)

chuckwagoncomputing, 2024-03-12:
Fixed with #6205 and https://github.com/rusefi/fw-custom-paralela/pull/5
Example commit: https://github.com/rusefi/fw-custom-paralela/commit/de776f85729688981fd2e282de90311d378a7ade

rusefillc, 2024-03-12:
https://github.com/rusefi/fw-custom-paralela/blob/master/generated/simulator_tune.msq says ``<constant digits="3" name="displacement" units="L">2.0</constant>`` while https://github.com/rusefi/fw-custom-paralela/blob/master/default_tune.cpp is asking for 1.8

rusefillc, 2024-03-17:
@chuckwagoncomputing seems not to work at https://github.com/rusefi/fw-custom-example could be user (me) issue?

Expected tune change did not happen after tune change https://github.com/rusefi/fw-custom-example/commit/c3b254d620663c6ecfed67c462ac8251697068ee

https://github.com/rusefi/fw-custom-example/ is green and produces tunes https://github.com/rusefi/fw-custom-example/commit/e82e6a66b07c7ee879b845aa94bd0e2f2281225c

PR #6216 (2024-03-15): Include board.mk in Simulator build if BOARD_DIR is set
Fix #6195
Green custom: https://github.com/chuckwagoncomputing/fw-custom-paralela/actions/runs/8289357534
Result: https://github.com/chuckwagoncomputing/fw-custom-paralela/commit/4aff60dc17f0982c33b50aff5956b874a1949f85
PR #6250 (2024-03-18): Fix custom sim
Fix #6195
Fix #6225
Green paralela: https://github.com/chuckwagoncomputing/fw-custom-paralela/actions/runs/8334784961
Green example: https://github.com/chuckwagoncomputing/fw-custom-example/actions/runs/8334897256
closed 2024-03-18 by rusefillc