rusefi/rusefi #6213: Realdash unreliable specifically over BT

rusefillc, 2024-03-14:
Realdash keeps disconnecting over BT

Magneelektronics, 2024-03-16:
stability of Bluetooth communication with rusEFI, I noticed that realdash has no problem with Bluetooth communication on Windows 10, but on Android the communication is unstable, has anyone tested communication with rusEFI via Bluetooth?
[image]

Automate1, 2025-09-30:
I would suspect its because of cheap chinese ELM327 clones.  Try a good quality one like from OBDLink.  See this from the AndrOBD software that has a setting to disable the ATZ command.

[image]