rusefi/rusefi #6214: Unit tests sometimes hang

chuckwagoncomputing, 2024-03-14:
Sometimes unit tests print `AddressSanitizer:DEADLYSIGNAL` repeatedly. This is more likely to happen when running sharded tests, though I think that's just because there are many more chances for it to occur. You can also get it to occur running rusefi_test directly.

Inserting `return 0;` as the first line of `main()` in unit_tests/main.cpp does not help, it still does it.

chuckwagoncomputing, 2024-03-15:
I have bisected this back to where asan was enabled in c72bd0717928513bbe3ee003990631d78bef3b2b, so that's no help.

rusefillc, 2024-03-15:
> I have bisected this back to where asan was enabled in [c72bd07](https://github.com/rusefi/rusefi/commit/c72bd0717928513bbe3ee003990631d78bef3b2b), so that's no help.

How? @chuckwagoncomputing are you able to reproduce locally?

mck1117, 2024-03-15:
FOME has started seeing the same thing in the last couple days, also always during the sharded run. Never seen an issue in the non-sharded run, or running locally.

chuckwagoncomputing, 2024-03-15:
Running on the rusefi-ci docker image, it happens about 1/3 of the time. Running on a GH hosted runner it happens much less frequently, apparently more on the order of 1/1000 because sometimes the sharded run with 600 instances completes successfully. I assume that it has the same cause, but I guess I don't know that for sure.

chuckwagoncomputing, 2024-03-15:
This is tough to diagnose because I haven't figured out how to get AddressSanitizer to dump a trace so I have no idea where it's happening. Also, AddressSanitizer doesn't work in GDB so that doesn't help either.

mck1117, 2024-03-15:
This gets weirder, [on this build](https://github.com/FOME-Tech/fome-fw/actions/runs/8261253127/job/22598196262), it's stuck with the yellow spinner, but if you click the gear and download the log archive, the run completed successfully?

chuckwagoncomputing, 2024-03-15:
If I download the archive, it only has the macos log. Maybe that's what you're seeing?

chuckwagoncomputing, 2024-03-16:
Where I have been able to reproduce this, or not:
- [ ] Archlinux Intel (GCC 13)
- [ ] Archlinux AMD (GCC 13)
- [ ] Archlinux GCC 11
- [ ] Archlinux Docker (GCC 13)
- [X] Ubuntu 22.04 Docker (GCC 11)
- [X] Ubuntu 23.10 Docker (GCC 13)
- [ ] Ubuntu 24.04 Docker (GCC 13)

Apparently something has fixed it in a newer version of some software.

mck1117, 2024-03-16:
This has to be a GHA thing, I poked an old commit from a few days ago and it failed in the new way: https://github.com/FOME-Tech/fome-fw/actions/runs/8305375089/job/22732190382

rusefillc, 2024-09-11:
is this still an issue?