rusefi/rusefi #6222: STFT deadband based on RPM

rusefillc, 2024-03-15:
on idle with shorter injection pulse we want wider deadband like 0.4

on higher RPM we can go towards 0.1

mck1117, 2024-03-15:
I've seen best results with _no_ deadband at all.

ElDominio, 2024-03-15:
Understood and agreed.

I should change the time setting between idle and main states to compensate for transport delays.

How does the decision to add/remove fuel work if there is no user-facing PID? I know the time thing setting affects how fast/aggressive the correction is, but,  how is the decision to remove more/less fuel made?

mck1117, 2024-03-15:
There kind of is PID actually, but it's only I, and that's what the time constant configures (but it's roughly the reciprocal of I gain).

mck1117, 2024-03-15:
That is to say, it's always adjusting (at 200hz or whatever), but the rate at which it adjusts is controlled by the time constant.

ElDominio, 2024-03-15:
Why are P and D not important? Legit question lol

P would affect how much to correct if I'm too far from my target (so make the trim grow quickly if target is 14 and say, were at 10), and D if we're too close could counteract P to smooth the transition from what I understand. Are these baked in or something?