rusefi/rusefi #6223: LTFT needs to exist to compensate for poorly running vehicles

ElDominio, 2024-03-15:
LTFT would allow to keep STFT in memory, then add additional STFT on top to be able to run a car with a bad fuel system

ElDominio, 2024-08-31:
Will add more details later but: LTFT should be stored in some permanent storage say every 5 seconds? Use SD as permanent storage?

ElDominio, 2024-08-31:
So about LTFT

We already have a decent STFT, so what we need LTFT to do is to take STFT and store it semi-permanently over time. The rate at which we save STFT as LTFT over time should be configurable, and we should also have a STFT/dt to consider STFT stable enough to be saved as "permanent".

We basically need to have a sum over time of STFT to normalize variations, then take that sum and add it to LTFT. For example, if we have LTFT jumping between 10-15, we should be average that over say, 5 seconds, then consider the result as LTFT (which would be 12.5 if we were perfectly jumping between our limits).

Other details:
We also need a button to reset LTFT 
We need LTFT per-bank for split bank engines
LTFT should also be saved in it's individual operating regions, (main, idle, wot, overrun)

benassi124, 2024-09-02:
My .02 regarding LTFT

- I think there should be a min engine run time before LTFT changes start to happen
- Ideally for load axis to match main VE table load axis.
-  An area where we can specify the speed of the correction in the different load area, it is ideal able to use a small percent of allowed correction at low load conditions where the engine spends most of the time, and faster at high loads where the engine spends less time.

A table with LTFTs would be be nice and the ability to apply the tables to ve or reset them, applying trim values should automatically reset the table, see the image below from maxxecu as an example
[image]

rusefillc, 2025-07-06:
preparation epic https://github.com/rusefi/rusefi/issues/8219

rusefillc, 2025-07-28:
@ggurov has shared

https://github.com/user-attachments/assets/0ee23020-ec57-4caa-8974-e1eed2583196

rusefillc, 2025-10-21:
We assume that it's done