rusefi/rusefi #6224: Brake input should be allowed to be shared between Launch Control and Misc

ElDominio, 2024-03-15:
**Is your feature request related to a problem? Please describe.**
I am always frustrated when I set my brake input in Misc settings and I then cannot use it for Launch Control since it creates a setting conflict

**Would we need new configuration fields?**
Why is Launch Input pin not part of misc settings? You should set the physical input under Misc settings, then use said input inside the Launch Control dialog box, with no configuration inside the Launch Control diagram since it is already configured under Misc.

**Describe the solution you'd like**
Remove pin setup from Launch, only choose if brake/clutch/button activate it, configured on misc.

highspeed427, 2024-06-21:
@rusefillc Where are we on this? 

thanks

rusefillc, 2024-06-23:
@kifir23917 I assume https://github.com/rusefi/rusefi/pull/6646 is fixing it?

kifir23917, 2024-06-24:
I would say that this issue relates to #6649, but I'm not sure that I understand what is a "setting conflict" mentioned in the description.

kifir23917, 2024-06-24:
[image]

kifir23917, 2024-06-24:
We decided that this problem is fixed with https://github.com/rusefi/rusefi/pull/6646

closed 2024-06-24 by rusefillc