rusefi/rusefi #6225: custom board simulator does not account prepend.txt

rusefillc, 2024-03-15:
https://github.com/rusefi/fw-custom-paralela/commit/d093ae70b25d4974b969a53f17b241ec65f8e654 is red because simulator seems to be creating binary tune snapshot using default value not custom ``LUA_SCRIPT_SIZE`` value which increases calibration image size

chuckwagoncomputing, 2024-03-15:
It seems like the value set by prepend.txt is being overwritten by the value from rusefi_config.txt which is read later. Has overriding rusefi_config.txt values with prepend.txt been known to work?

rusefillc, 2024-03-15:
> It seems like the value set by prepend.txt is being overwritten by the value from rusefi_config.txt which is read later. 

100%

proteus https://github.com/rusefi/rusefi/blob/master/firmware/config/boards/proteus/prepend.txt has size override which 500% make it into firmware and java console

PR #6250 (2024-03-18): Fix custom sim
Fix #6195
Fix #6225
Green paralela: https://github.com/chuckwagoncomputing/fw-custom-paralela/actions/runs/8334784961
Green example: https://github.com/chuckwagoncomputing/fw-custom-example/actions/runs/8334897256
closed 2024-03-18 by rusefillc