rusefi/rusefi #6247: "Switch to OpenBLT" command doesn't work on Proteus-F7

andreika-git, 2024-03-18:
The "reboot_openblt" command reboots the ECU, but OpenBLT doesn't switch to "backdoor" mode on Proteus-F7...
Something around SharedParamsReadByIndex(0, &value): the value != 1 (but the function returns BLT_TRUE)
The same build works on 2chan (recompiled locally or from bundle).

mck1117, 2024-03-18:
flush the cache before reboot

rusefillc, 2024-03-18:
> flush the cache before reboot

@mck1117 do you have a relevant FOME line? How does this work for you on F7?

dron0gus, 2024-03-24:
https://github.com/rusefi/rusefi/pull/6291

mck1117, 2024-03-24:
> @mck1117 do you have a relevant FOME line? How does this work for you on F7?

it works and I'm not sure why

andreika-git, 2024-03-26:
Fixed by https://github.com/rusefi/rusefi/commit/4c7cd85f8bf7b8f6137af45f45c09b30b0dc51fe

closed 2024-03-26 by andreika-git