rusefi/rusefi #6253: Cypress and Kinetis board-specific firmware fails to build

chuckwagoncomputing, 2024-03-19:
The Cypress and Kinetis config directories aren't included in the simulator build. Before we were just setting SHORT_BOARD_NAME to f407-discovery, but now we have to set SHORT_BOARD_NAME to be board-specific if we want to include the info from prepend.txt for that board. The generated headers for Cypress and Kinetis go in their own directories, not with the rest of the boards. The makefile adds META_OUTPUT_ROOT_FOLDER/controllers/generated to the include list. META_OUTPUT_ROOT_FOLDER could be added to the meta-info files for Cypress and Kinetis, but their generated .h files are at controllers/algo instead of controllers/generated. Would it be acceptable to move them?

chuckwagoncomputing, 2024-03-19:
status update: https://github.com/chuckwagoncomputing/rusefi/actions/runs/8336663644

rusefillc, 2024-03-19:
next issue: rusefi_generated_cypress.h seems to be stale since March 8th regardless of the move https://github.com/rusefi/rusefi/pull/6260

chuckwagoncomputing, 2024-03-20:
This approach has a problem. The .ini file ends up in config/boards/cypress/config/tunerstudio/

rusefillc, 2024-03-20:
> This approach has a problem. The .ini file ends up in config/boards/cypress/config/tunerstudio/

if all the scripts pick those .ini from config/boards/cypress/config/tunerstudio/ I do not see a problem

chuckwagoncomputing, 2024-03-20:
What's worse, it puts all the generated enums there too. That is a problem, because both the header in firmware/controllers and the one in firmware/config/board/cypress/controllers get included and give multiple definition errors.

If there weren't non-generated files in those directories, we could just not have them as an included directory, but unfortunately the generated and non-generated files are in the same directories.

chuckwagoncomputing, 2024-03-20:
Why are Cypress, Kinetis, and Subaru_EG33 weird in the first place? Why can't they have their config files in the normal place like normal folk?

chuckwagoncomputing, 2024-03-20:
FOME has moved all generated files into a single dir: https://github.com/FOME-Tech/fome-fw/pull/348
I think we can do things a little better by assuming META_OUTPUT_ROOT_FOLDER to be something like "generated" if it is empty. The result would be a "shadow hierarchy" matching the current one, but only including generated files.
Then we would just need to add all the directories in the shadow hierarchy to the includes list.
i.e.:
```
$(META_OUTPUT_ROOT_FOLDER)console/binary
$(META_OUTPUT_ROOT_FOLDER)controllers/actuators
$(META_OUTPUT_ROOT_FOLDER)controllers/algo
$(META_OUTPUT_ROOT_FOLDER)controllers/engine_cycle
$(META_OUTPUT_ROOT_FOLDER)controllers/math
$(META_OUTPUT_ROOT_FOLDER)controllers/sensors
$(META_OUTPUT_ROOT_FOLDER)controllers/tcu
$(META_OUTPUT_ROOT_FOLDER)controllers/trigger
$(META_OUTPUT_ROOT_FOLDER)unit_tests/test-framework/
```

rusefillc, 2024-03-20:
> Why are Cypress, Kinetis, and Subaru_EG33 weird in the first place? Why can't they have their config files in the normal place like normal folk?

Those three have their own version of rusefi_hw_enums.h

At the moment in order to use custom rusefi_hw_enums.h in light of small variations of two ``-enumInputFile`` arguments vs one ``-enumInputFile`` arguments we have the questionable copy-paste

@chuckwagoncomputing see https://github.com/rusefi/rusefi/commit/1dce194f122dabb60f3669ea6246dca9e7f008da

@chuckwagoncomputing my preference is to go forward with smaller safer steps. Before we go change normal firmware first let's please fix those three exceptions.

@chuckwagoncomputing once we fix I agree that we shall unify

once everything green we can refactor. let's please separate refactoring from repair work

@chuckwagoncomputing is get_subaru_config.sh simply not used at the moment?

chuckwagoncomputing, 2024-03-20:
gen_subaru_config.sh is used, I just didn't realize it was broken when I made this issue.

rusefillc, 2024-03-20:
> gen_subaru_config.sh is used, I just didn't realize it was broken when I made this issue.

ohh I see it now ``CUSTOM_GEN_CONFIG=config/gen_subaru_config.sh``

chuckwagoncomputing, 2024-03-20:
The only way I can think to fix this without touching anything else is to make cypress, kinetis, and subaru put their generated files in the same place all the other boards do, but then we have to be careful to not push those to git.

chuckwagoncomputing, 2024-03-21:
FYI kinetis is failing now for a different reason:
```
In file included from ./hw_layer/ports/rusefi_halconf.h:8,
                 from ././config/boards/kinetis/halconf.h:31,
                 from ChibiOS/os/hal/include/hal.h:30,
                 from ./global.h:15,
                 from ./pch/pch.h:21:
././config/boards/kinetis/board_configuration.cpp: In function 'void setSerialConfigurationOverrides()':
././config/boards/kinetis/efifeatures.h:267:22: error: unsigned conversion from 'int' to 'uint16_t' {aka 'short unsigned int'} changes value from '115200' to '49664' [-Werror=overflow]
  267 | #define SERIAL_SPEED 115200
      |                      ^~~~~~
././config/boards/kinetis/board_configuration.cpp:36:55: note: in expansion of macro 'SERIAL_SPEED'
   36 |         engineConfiguration->tunerStudioSerialSpeed = SERIAL_SPEED;
      |                                                       ^~~~~~~~~~~~
```

rusefillc, 2024-03-21:
> FYI kinetis is failing now for a different reason

my bad https://github.com/rusefi/rusefi/commit/b88a176f350f7732b79356aae41856053fc2cc38

closed 2024-03-24 by chuckwagoncomputing