rusefi/rusefi #6261: stm32f767_nucleo status

rusefillc, 2024-03-19:
```
** Programming Started **
Info : device id = 0x10016451
Info : flash size = 2048 KiB
Info : Single Bank 2048 kiB STM32F76x/77x found
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
in procedure 'program'
hint: error state is already in your clipboard, please use PASTE or Ctrl-V while reporting issues
!!! FIRMWARE FLASH: DOES NOT LOOK RIGHT !!!
```

rusefillc, 2024-03-19:
```
2024-03-19_13_31_22_340: EngineState: *** detected HSE clock 0.00 MHz PLLM = 8
2024-03-19_13_31_22_342: EngineState: hellenBoardId=0
2024-03-19_13_31_22_343: EngineState: UID=240031 32325104 33393832
2024-03-19_13_31_22_344: EngineState: MCU rev=451 flashSize=2048
```

x893, 2024-11-19:
Hello,
add one line in file `firmware/hw_layer/ports/stm32/osc_detector.cpp` at line 179

```
A	float hseFrequencyHz = 10 * rtcpreDivider * STM32_TIMCLK2 / hseCounts;
d-->	hseFrequencyMhz = hseFrequencyHz / 1e6;
d	autoDetectedRoundedMhz = ((int)hseFrequencyHz + (1e6 / 2)) / 1e6;

```

rusefillc, 2024-11-19:
> Hello, add one line in file `firmware/hw_layer/ports/stm32/osc_detector.cpp` at line 179

thank you! https://github.com/rusefi/rusefi/commit/0328c08f2b8a46e9c9be9aa70855130eec6088b3

rusefillc, 2024-11-19:
Totally not clear what this ticket was originally about, looks like stm32f767_nucleo more or less works?

closing

closed 2024-11-19 by rusefillc