rusefi/rusefi #6262: Alpha-n leaves both ve and ign table with kpa(map) y axis

runesmdk, 2024-03-19:
As title. March 740 firmware

rusefillc, 2024-03-19:
@runesmdk please provide more details. what exactly is desired behavior?

please note MAP estimation table and functionality it could be relevant

runesmdk, 2024-03-20:
Desired behavior is y axis is tps in relevant tables like ve, afr and selectable on ign table to be map or tps. 

Map estimation sounds like a hard time to get decent before then tuning map ve, specially because of the non linearity of map signal on itbs vs rpm.

mck1117, 2024-03-20:
the label is wrong in override mode, it works as tps but just says kpa

runesmdk, 2024-03-20:
Okay. In that case, i dont know what the map estimation table is for? 

Also, wont the wrong labels break autotune functionality?

runesmdk, 2024-03-20:
Ah yea, autotune works, but megalogviewer ve analyze doesnt. I had the issue while trying to dial in my itb's on a speeduino. And that was fixed with proper labeling.

rusefillc, 2024-03-20:
@runesmdk could https://github.com/rusefi/rusefi/issues/6264 be relevant?

runesmdk, 2024-03-20:
> @runesmdk could #6264 be relevant?

As a hack yea. But it would be better for it to have ve table use tps when on alpha n, so it would just work as is. 

And for ignition table to have its own definable y axis (map or tps)