rusefi/rusefi #6281: bosch quick start decoding issue or else

rusefillc, 2024-03-24:
Feb FW good
[image]

current FW same HW same vehicle bad
[image]

rusefillc, 2024-03-24:
Depends on https://github.com/rusefi/rusefi/issues/6285

andreika-git, 2024-03-24:
Proteus-F7 artefacts from 2024-02-29 to 2024-03-04 (from older to later):
* https://github.com/rusefi/rusefi/commit/af81c460499e6a3e27c79b913e5fd7bbac6e3168 https://github.com/rusefi/rusefi/actions/runs/8088622645/artifacts/1284232708
...
* https://github.com/rusefi/rusefi/commit/7207635602fe36ca06d3ca71e406edc8a94f72b4 https://github.com/rusefi/rusefi/actions/runs/8133769005/artifacts/1293321219

rusefillc, 2024-03-24:
https://github.com/rusefi/rusefi/commit/d3461e39051b070adeeb451d64a036a062cbc01f has messed up either cranking or something

rusefillc, 2024-03-24:
feb 25 good
[image]

match 1st problems
[image]

rusefillc, 2024-03-25:
See https://github.com/rusefi/rusefi/pull/6296

rusefillc, 2024-03-25:
[feb-25-firmware-started.zip](https://github.com/rusefi/rusefi/files/14746231/feb-25-firmware-started.zip)