rusefi/rusefi #6282: custom board should be able to add unit tests

rusefillc, 2024-03-24:
custom boards like https://github.com/rusefi/fw-custom-paralela could have custom logic

custom logic would need custom units tests, would be great to reuse existing unit test infrastructure just plug extra tests

rusefillc, 2024-03-24:
@chuckwagoncomputing any chance you can help?

chuckwagoncomputing, 2024-03-24:
Do you thing the unit tests should have their own separate workflow, or be a part of the existing custom board build action?

rusefillc, 2024-03-24:
> Do you think the unit tests should have their own separate workflow, or be a part of the existing custom board build action?

divide and conquer is asking for separate but I do not have a very strong opinion, we can start as part of existing if that's easier.

rusefillc, 2024-05-16:
@chuckwagoncomputing poke... if you can help

rusefillc, 2024-09-11:
We seem to have this now!