rusefi/rusefi #6285: VVT simulator on real hardware pin

rusefillc, 2024-03-24:
We have VVT https://github.com/rusefi/rusefi/issues/5640 in simulator

Next step would be to produce VVT signal alongside crank signal on real hardware

Specifically it's desired to have 60/2 on crank PD0 and Bosch Quick Start on CAM/VVT on PD1 on discovery

rusefillc, 2024-03-24:
configuration pin added https://github.com/rusefi/rusefi/commit/a309e9f3663008b769581d870f429ddd05366673

OutputPin wiring attempt https://github.com/rusefi/rusefi/commit/4d1b59d15780f529f685ca551f671ad5b308f644

but logic analyzer shows generated 60/2 crank but nothing on VVT generated

andreika-git, 2024-03-24:
Well, it works for me (compiled from current master, no changes)...

[image]

[image]
[image]

rusefillc, 2024-03-27:
cam pin output mode added

closed 2024-03-27 by rusefillc