rusefi/rusefi #6286: nighly releases have stopped on March 1st

rusefillc, 2024-03-24:
https://github.com/rusefi/rusefi/releases

chuckwagoncomputing, 2024-03-24:
No changes to build-firmware.sh happened then. This makes me think that it is more likely something wrong with the token. Did you generate a new token or something?

rusefillc, 2024-03-24:
@chuckwagoncomputing relevant runs at at https://github.com/rusefi/rusefi/actions/workflows/build-firmware.yaml?query=event%3Aschedule right?

I see same 'ACCESS_TOKEN' has two usages - nightly releases and wiki sync, let me have a look at wiki sync status. Later on I would also probably split into two separate tokens on maybe just rename to make it a bit easier to reverse engineer later :)

rusefillc, 2024-03-24:
@chuckwagoncomputing I see recent "merge rusEFI wiki" at https://github.com/rusefi/rusefi_documentation/commits/master/ and at the moment my understanding is that same token is used for that?

rusefillc, 2024-03-24:
https://github.com/mathieudutour/github-tag-action/issues/165 seems to agree that it's a lack of permission situation

rusefillc, 2024-03-24:
@chuckwagoncomputing I see that https://github.com/mathieudutour/github-tag-action?tab=readme-ov-file#-inputs says "Usually ${{ secrets.GITHUB_TOKEN }}." do you remember why we do not use it?

Anyway with https://github.com/rusefi/rusefi/commit/6cd1e39df55124246402de4ca0d2c1b3884f3355 change the error message has changed to "Resource not accessible by integration" 

What do we take from the fact that error is different but still error?

Random google link https://github.com/orgs/community/discussions/60820

chuckwagoncomputing, 2024-03-25:
I managed to make this work with a fine-grained token with everything set to read/write that I could, now to figure out which one did it...

rusefillc, 2024-03-26:
Worked with crazy scope set on NIGHTLY_RELEASE_ACCESS_TOKEN will now reduce it

https://stackoverflow.com/questions/63906613/minimal-set-of-scopes-to-push-to-github-using-an-access-token says "As per JetBrains the Minimal set of scopes which are require are basically -> repo, workflow, read:org, gist."

rusefillc, 2024-03-26:
Just removed

[image]

We shall see how it works tomorrow

chuckwagoncomputing, 2024-03-26:
I've been testing too, with a fined-grained token. Here are my current findings:
Doesn't work with:
Actions, Administration, Contents, Deployment
Works with:
Actions, Administration, Contents, Deployment, Custom Properties, Webhooks, Workflows

So at least one of Custom Properties, Webhooks, Workflows made it work.

I'm not sure how these relate to the classic token permissions.

chuckwagoncomputing, 2024-03-26:
I propose that I try Custom Properties and you try Webhooks.
If both fail, most likely Workflows is the one that is needed.

rusefillc, 2024-03-26:
> I propose that I try Custom Properties and you try Webhooks. If both fail, most likely Workflows is the one that is needed.

to make tests cleaner I will change after it attempts tomorrow

chuckwagoncomputing, 2024-03-26:
It's based on UTC, so "tomorrow" is in 4 minutes.

chuckwagoncomputing, 2024-03-27:
Looks like "Workflows" is the secret sauce, I just had a successful run with only "Contents" and "Workflows"