rusefi/rusefi #6292: uaefi stops firing ignition correcty after running for some time at high rpm

ggurov, 2024-03-24:
**Describe the bug**
unit was on a bench, at 7000 rpm, after some time it was noticed that ignition status led were not blinking the right pattern
logic analyzer used to calture pattern

[image]

after reset (power down 12v and pull usb, nust 12v power down did now fix it)

the pattern went back to proper 6 cyl 
[image]
pattern

**Hardware**
 - uaefi
 - bench with 36-2 + cam stim

**To Reproduce**
Steps to reproduce the behavior:
1. set 7000 rpm on stim
2. lock timing to 4 degrees btdc
3. wait a while ?
4. maybe see bad pattern ?

**Expected behavior**
ignition to continue expected pattern

**Additional context**
i don't think i'm crazy

ggurov, 2024-03-26:
confirmed not crazy

unit uptime: 148000 seconds, 41 hours give or take. was fine at 35-36 hours

uaefi rusefi 20240312

tune: https://rusefi.com/online/view.php?msq=1629
short log: https://rusefi.com/online/view.php?log=1362

reproduce this again, almost 48 hours of unit being on ~10000 rpm
locked ignition timing,, not any particular value

1000 rpm logic analyzer capture:
console indicates 4ms dwell, there's at least 1 of those present in one of the ignition channels (D4-D9)
injection doesn't seem to be affected (D10-D15)

[image]

this seems to still be going up:

[image]

console seems to throw an exception:

[image]

[image]

the pattern changes with RPM 2700 wavey

[image]

2500 rpm,, clusters:

[image]

sometimes there's signals absent in channels completely:

[image]

console engine sniffer shows weird coil durations:

[image]

longer chart

[image]

2024-03-26_06_44_11_977: EngineState: WARNING: C9009: looks like skipped spark event 11952948 Coil 3
2024-03-26_06_44_11_978: EngineState: WARNING: C9011: re-adding element into event_queue

2024-03-26_06_48_05_120: EngineState: WARNING: C9009: looks like skipped spark event 11956691 Coil 5
2024-03-26_06_48_05_123: EngineState: WARNING: C9011: re-adding element into event_queue

[image]

these warnings seem to be accumulating quite rapidly

	CUSTOM_OUT_OF_ORDER_COIL = 9012,

after fiddling with nothing much really, pattern changed to this:

half the ignition channels at like 1/10th dwell, for a few seconds, then switch to the other 3 channels with the same- 2500 rpm

[image]

around 7500 rpm, dwell starts out from 0, increases to indicated 4ms, then disappears:

[image]

sometimes channels seem to fire together: 2 5 on the left, and 3-4 on the right

[image]

3600 rpm:

no signal at all on ign1 or ign6, which after a few seconds flips to other channels being empty.

[image]

rusefillc, 2024-03-26:
Thank you @ggurov this is valuable data! Let me add some tools to reproduce this first.

ggurov, 2024-03-26:
https://rusefi.com/online/view.php?log=1363 better log file, previous log file has time jumping by quarter of a second while garbage collection runs, this one is clean.

rusefillc, 2024-03-26:
@ggurov any chance you can attempt reproducing same with self-stimulation?

@ggurov any chance you could attempt same with lower tooth trigger count?

@ggurov any chance you could attempt this with three times slower RPM?

mck1117, 2024-04-09:
48 hours at 10k rpm on a 36-1 is almost exactly 2^30 teeth

mck1117, 2024-04-09:
I think the tl;dr might be "don't run for that long"

rusefillc, 2024-04-09:
> what about generators ? they run longer

+1

rusefillc, 2024-05-03:
@kifir23917 
* what's the smart way to approach this issue?
* would a unit test with very quick time movement help us here?

rusefillc, 2024-05-12:
Wonder if this ticket here is a variation of #6451? Need to reproduce with current firmware.

rusefillc, 2024-05-14:
Closing under the assumption of dup of https://github.com/rusefi/rusefi/issues/6451

closed 2024-05-14 by rusefillc