rusefi/rusefi #6324: Jeep XJ 4 Cyl not on "All Supported Triggers"  page

ElDominio, 2024-03-31:
**Describe the bug**
There is no description of "Jeep XJ 4cyl"  on https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers#jeep page

**Hardware**
 - Board: 2chan 
 - Engine info Jeep 2.5L engine

**Expected behavior**
There is some sort of visualization of what this trigger expects to see, like the following screenshot:

**Screenshots**
[image]

Is it the upper one that is expected on Jeep XJ 4 cyl?

ElDominio, 2024-03-31:
[image]

I'm working on a setup with this trigger, but I see it wants primary/secondary channel even though they are split from the factory? Shouldn't this do crank decoding seperate from cam decoding, and shouldn't it just use single tooth cam?

rusefillc, 2024-03-31:
@ElDominio https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers#jeep-4-cyl added

rusefillc, 2024-03-31:
> Shouldn't this do crank decoding separate from cam decoding, and shouldn't it just use single tooth cam?

yes it should

closed 2024-04-06 by ElDominio