rusefi/rusefi #6333: six generated files not pushed into git

rusefillc, 2024-04-03:
i would expect these files to be auto-generated by actions and not be modified locally, i believe that was the case at some point?

[image]

rusefillc, 2024-04-03:
master is red with
```
Run git config --local user.email "action@github.com"
fatal: pathspec 'console/binary/generated/*.ini' did not match any files
```

chuckwagoncomputing, 2024-04-03:
I think it needs a leading firmware/ but I can't get to it until late tonight.

rusefillc, 2024-04-03:
> I think it needs a leading firmware/ but I can't get to it until late tonight.

Done https://github.com/rusefi/rusefi/commit/2cea463027cc8797087ccaca35f99edac611794e thank you @chuckwagoncomputing

PR #6334 (2024-04-03): add missing .ini and .md files
This should fix #6333
closed 2024-04-03 by rusefillc