rusefi/rusefi #6337: cd simulator ; make always full build

rusefillc, 2024-04-04:
cd simulator 
make
make

expected: full and incremental build
actual result: two full builds

chuckwagoncomputing, 2024-04-04:
Could not reproduce on current master. Any more details? I assume you mean with no changes between the two invocations of `make`.

rusefillc, 2024-04-04:
```
rusefi@DESKTOP-57S88S8:~/rusefi/simulator$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
rusefi@DESKTOP-57S88S8:~/rusefi/simulator$ make -j12
```

```
simulator/boards.cpp: In function ‘int getAdcValue(const char*, int)’:
simulator/boards.cpp:14:29: warning: unused parameter ‘msg’ [-Wunused-parameter]
   14 | int getAdcValue(const char *msg, int hwChannel) {
      |                 ~~~~~~~~~~~~^~~
simulator/boards.cpp:14:38: warning: unused parameter ‘hwChannel’ [-Wunused-parameter]
   14 | int getAdcValue(const char *msg, int hwChannel) {
      |                                  ~~~~^~~~~~~~~
Compiling util_dummy.cpp
Compiling crc.cpp
Compiling efistringutil.cpp
Compiling fragments.cpp
Compiling math.cpp
Compiling pt2001.cpp
Compiling main.cpp
Linking build/rusefi_simulator
rusefi@DESKTOP-57S88S8:~/rusefi/simulator$ git status
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        modified:   ../firmware/controllers/generated/rusefi_generated_f407-discovery.h
        modified:   ../firmware/controllers/generated/signature_f407-discovery.h
        modified:   ../firmware/tunerstudio/generated/rusefi_f407-discovery.ini
        modified:   ../java_console/models/src/main/java/com/rusefi/config/generated/Fields.java

no changes added to commit (use "git add" and/or "git commit -a")
rusefi@DESKTOP-57S88S8:~/rusefi/simulator$ git status ; make -j12
```

expected: not much with second ``make``

actual result:
```
simulator/boards.cpp: In function ‘int getAdcValue(const char*, int)’:
simulator/boards.cpp:14:29: warning: unused parameter ‘msg’ [-Wunused-parameter]
   14 | int getAdcValue(const char *msg, int hwChannel) {
      |                 ~~~~~~~~~~~~^~~
simulator/boards.cpp:14:38: warning: unused parameter ‘hwChannel’ [-Wunused-parameter]
   14 | int getAdcValue(const char *msg, int hwChannel) {
      |                                  ~~~~^~~~~~~~~
Compiling util_dummy.cpp
Compiling crc.cpp
Compiling efistringutil.cpp
Compiling fragments.cpp
Compiling math.cpp
Compiling pt2001.cpp
Compiling main.cpp
Linking build/rusefi_simulator
rusefi@DESKTOP-57S88S8:~/rusefi/simulator$ git status
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        modified:   ../firmware/controllers/generated/rusefi_generated_f407-discovery.h
        modified:   ../firmware/controllers/generated/signature_f407-discovery.h
        modified:   ../firmware/tunerstudio/generated/rusefi_f407-discovery.ini
        modified:   ../java_console/models/src/main/java/com/rusefi/config/generated/Fields.java

no changes added to commit (use "git add" and/or "git commit -a")
rusefi@DESKTOP-57S88S8:~/rusefi/simulator$ git status ; make -j12
```

rusefillc, 2024-05-16:
Poke @chuckwagoncomputing ... Pretty please?

rusefillc, 2024-05-16:
Actually not an issue on Windows

[image]

rusefillc, 2024-06-06:
still works on Windows
[image]

rusefillc, 2024-06-06:
By the way ```cd simulator
make``` produces only one version of simulator as opposed to https://github.com/rusefi/rusefi/issues/6585

[image]

rusefillc, 2024-09-12:
```
   Prerequisite '.os-sentinel' of target 'build/obj/hal_st.o' does not exist.
  Must remake target 'build/obj/hal_st.o'.
```

rusefillc, 2024-09-12:
See https://github.com/rusefi/rusefi/commit/4ea3cc868c5a29faa98056d1586fa1dcb6552c0c

Looks like we have a feature to specify target OS and we fails to handle when that argument is not specified? or just fail to handle not-windows?

```
$(OBJS): .os-sentinel

.os-sentinel: .FORCE
	if [ "$$(cat $@ 2>/dev/null)" != $(OS) ]; then \
	echo $(OS) >$@; fi
```

rusefillc, 2024-09-12:
PS: maybe that's rebuild mechanism for building linux simulator after Windows simulator build on Linux? keywords 

```
# Custom board builds don't include the simulator
ifneq ($(BUNDLE_SIMULATOR),false)
  SIMULATOR_EXE = ../simulator/build/rusefi_simulator.exe
endif
```

```
# make sure not to invoke in parallel with SIMULATOR_EXE rule above
../simulator/build/rusefi_simulator.linux: $(CONFIG_FILES) .FORCE
	$(MAKE) -C ../simulator -r OS="Linux" SUBMAKE=yes

# make Windows simulator a prerequisite so that we don't try compiling them concurrently
# that also means no incremental compilation making that rule less useful. See 'rusefi_simulator.linux' above
../simulator/build/rusefi_simulator.both: $(CONFIG_FILES) .FORCE | $(SIMULATOR_EXE)
	$(MAKE) -C ../simulator -r OS="Linux" SUBMAKE=yes
```

closed 2024-09-14 by rusefillc