rusefi/rusefi #6349: Overdwell at times around syncloss while cranking

ggurov, 2024-04-09:
**Describe the bug**
At times while cranking when sync loss happens, , there is long dwell periods, which could damage components

**Hardware**
 - Board: uaefi
 - Engine : bench

**To Reproduce**
Steps to reproduce the behavior:
1. set external trigger wheel to below cranking  rpm 
2. start logic analyzer capture
3. observe sync loss occur (if it does, for one reason or another)
4. examine signals

**Expected behavior**
Dwell not to exceed a safe-ish treshold

**Screenshots**
example of 150ms dwell

[image]

example of 460ms dwell:

[image]

rusefillc, 2024-04-09:
@ggurov please provide recipient rusEFI tune

@ggurov any chance of rusEFI logs for when the issue happens?

@ggurov what do you use for external signal simulation? would you be able to reproduce with external rusEFI maybe on discovery?

ggurov, 2024-04-09:
more info: sync loss while cranking came from noisy ADC that drove target RPM on a stim.

stim adjusted RPM between teeth while outputing signals:

[image]

so RPM changes rather wildly while virtual wheel is rotating, confusing decoder with long/short teeth.

this generated decent amount of sync loss. leading to the discovery of overdwell.

ggurov, 2024-04-09:
reproduced to capture tune and log:

tune: https://rusefi.com/online/view.php?msq=1637
log: https://rusefi.com/online/view.php?log=1375

relevant: 
[image]

coilstate3 should not be long like that.

rusefillc, 2024-04-09:
@ggurov just ordered https://www.amazon.com/gp/product/B0BC7FM7FD

ggurov, 2024-04-09:
external trigger is done using an arduino uno ( https://www.amazon.com/WWZMDiB-ATmega328P-microcontroller-Development-Compatible/dp/B0BV9XFQ5T/ - has to be atmega328 based ) and an optoisolator board (https://www.amazon.com/gp/product/B0BC7FM7FD) that's driving a 12v signal from UNO's outputs

software is "Ardustim", slightly modified to not need the UI - attached here

[ardustim-for-overdwell.zip](https://github.com/rusefi/rusefi/files/14923290/ardustim-for-overdwell.zip)

this version boots up with 36-2 + 1 cam hardcoded, storage.ino:

```
void loadConfig()
{
  if(EEPROM.read(EEPROM_VERSION) == 255 || 1)
  //if(true)
  {
    //New arduino
    selected_wheel = THIRTY_SIX_MINUS_ONE_WITH_SECOND_TRIGGER; //36-1  <-----------
//    selected_wheel = SIXTY_MINUS_TWO_WITH_CAM; //36-1
    wanted_rpm = 200;
    mode = POT_RPM;
  }

```

pin 8 is crank
pin 9 is cam

  ```
pinMode(8, OUTPUT); /* Primary (crank usually) output */
  pinMode(9, OUTPUT); /* Secondary (cam1 usually) output */
```

A0 input gets a variable resistor to adjust RPM.

using the optoisolator board:

arduino pin 8 goes to in1
arduino pin 9 goes to in2
join the grounds
HVCC goes to 12v

OUT1 goes to crank in on ECU - hall ( may need 1k pullup to 12v)
OUT2 goes to cam in on ECU - hall ( may need 1k pullup to 12v)

this simulates a 12v hall effect sensor, inversion is taken care of, what goes into IN, comes out of OUT logically

rusefillc, 2024-04-10:
@ggurov https://github.com/rusefi/Ardu-Stim/tree/30-to-300 with https://github.com/rusefi/Ardu-Stim/commit/8d08850ba7f3582188397a7a9dc6e79fc0dda2fe

mck1117, 2024-04-10:
is overdwell protection somehow broken?

ggurov, 2024-04-10:
there is no setting for overdwell

mck1117, 2024-04-10:
> there is no setting for overdwell

correct, it's supposed to be always enabled at 1.5x normal dwell

rusefillc, 2024-04-10:
> correct, it's supposed to be always enabled at 1.5x normal dwell

there is '!limitedSpark' check around there which I am not sure about

rusefillc, 2024-04-10:
@ggurov shot into the darkness can you try artifact from https://github.com/rusefi/rusefi/pull/6351?

rusefillc, 2024-04-10:
@ggurov master has some extra logging now https://github.com/rusefi/rusefi/commit/189539b51fb30bafc87f0c2c1005aa51f736bd72

rusefillc, 2024-04-10:
@ggurov any chance you can add a couple of photos of the complete test setup?

For some reason I am tempted to simply solder some of the revenant things together.

ggurov, 2024-04-10:
[image]
[image]

2024-04-10_11_45_41_840: EngineState: WARNING: C9003: primary trigger error: not enough teeth between sync points: expected 34/0 got 21/0
2024-04-10_11_45_43_935: EngineState: WARNING: C9012: out-of-order coil off Coil 3

[image]

ggurov, 2024-04-10:
@rusefillc 
there's not much there, here's the picture.

it's a fairly basic engine simulator, iat/clt, tps/o2/map (unlabeled)

relevant parts are the boards on the right, blue board with atmega328 in a zif socket is my arduino uno that's running the code.

from that arduino uno, pin8 goes to crank input, pin9 goes to cam input on uaefi through the mess of wires on the left at the bottom of the board with variable resistors

the variable resistor labeled RPM is hooked up to 5v, A0, and ground on arduino uno to adjust RPM.

should be a pretty minimal set of wires

to uaefi: 
ground
12v
 arduino pin8 to crank (vr or hall ?)
arduino pin9 to cam (vr or hall ?)
arduino ground

usbc

on arduino variable resistor ground, a0, aref

[image]

ggurov, 2024-04-10:
the on time for that channel (D6) lasts pretty much a full rev... give or take a few ms. 

[image]

rusefillc, 2024-04-11:
@ggurov fancy approach to investigate would be https://github.com/rusefi/rusefi/issues/6355

mck1117, 2024-04-12:
@ggurov do you intend to have `Always use instant RPM` enabled on the experimental page?

ggurov, 2024-04-12:
it makes the logs look nicer, if it’s better internally for running consistency to have that off, it wold be nice to have the logs be more smooth transition . i’ll give MLV’s smoothing a shot.

what’s the recommendation ? probably needs another thread so we don’t pollute this one.

rusefillc, 2024-04-12:
@mck1117 thank you!

@ggurov we have a chance can you please test?

ggurov, 2024-04-12:
i'll pull down artifacts from #6349 and try it out, but i just disabled instant RPM(reboot and powerdown and stuff proper), was able to get overdwell around cranking RPM with noisy trigger:

[image]

rusefillc, 2024-04-12:
> i just disabled instant RPM(reboot and powerdown and stuff proper), was able to get overdwell around cranking RPM with noisy trigger:

plz provide snapshots. First two are already in this repo just no decoding and no testing code so far https://github.com/rusefi/rusefi/commit/9c3385465371f3486492127e796a2c3a39dad4db

ggurov, 2024-04-12:
[2ch-overdwell-no-instant-rpm.zip](https://github.com/rusefi/rusefi/files/14961511/2ch-overdwell-no-instant-rpm.zip)

ggurov, 2024-04-12:
with https://github.com/rusefi/rusefi/commit/752db20ba2ff04c1f61906194d9011e3f8c1704a 

overdwell still reproduced with noisy trigger:

[image]

[1ch-overdwell-with-752db20-master-65034.zip](https://github.com/rusefi/rusefi/files/14961555/1ch-overdwell-with-752db20-master-65034.zip)

tune: https://rusefi.com/online/view.php?msq=1639

mck1117, 2024-04-12:
> with [752db20](https://github.com/rusefi/rusefi/commit/752db20ba2ff04c1f61906194d9011e3f8c1704a)
> 
> overdwell still reproduced with noisy trigger, instant rpm is off in the tune as well.

try it with instant RPM turned back on

ggurov, 2024-04-12:
this is the longest one i've seen so far: ign ch 6 (D9), on for 8.6 seconds, instant RPM ON

[image]

[ign-ch6-on-for-8-seconds.zip](https://github.com/rusefi/rusefi/files/14961590/ign-ch6-on-for-8-seconds.zip)

rusefillc, 2024-04-17:
@ggurov can you please test again in light of https://github.com/rusefi/rusefi/commit/4bc38d3543a8a9faf7416ecfeb7dff08a9dba86f?

ggurov, 2024-04-18:
looks much better, cannot reproduce the condition anymore back to back previous firmware -> master 20240418 👍

closed 2024-04-18 by rusefillc