rusefi/rusefi #6355: logic analyzer binary or CSV to reproduce Overdwell defect without real hardware

rusefillc, 2024-04-11:
See https://github.com/rusefi/rusefi/blob/master/unit_tests/tests/trigger/resources/readme.md

Would be great to have full logic analyzer recording of the incident which I would be able to convert into same or similar CSV format, and reproduce the issue without real hardware!

mck1117, 2024-04-11:
doesn't saleae let you save as a csv?

rusefillc, 2024-04-11:
> doesn't saleae let you save as a csv?

Yes if @ggurov uses a saleae we are golden. If @ggurov uses something else, that something else would need to provide same functionality. Ideally I need both a file working with UI (which for saleae means .logicdata) and CSV.

mck1117, 2024-04-11:
his screenshots are of sigrok, which I guarantee can export as csv as well

rusefillc, 2024-04-11:
> his screenshots are of sigrok, which I guarantee can export as csv as well

TIL https://sigrok.org/wiki/PulseView

ggurov, 2024-04-11:
attached are sigrok session files that contain runs with overdwell. 

[overdwell-sigrok-sessions.zip](https://github.com/rusefi/rusefi/files/14948359/overdwell-sigrok-sessions.zip)

rusefillc, 2024-04-12:
export takes FOREVER. it has moved from 0% but it plans to somehow spend a few MINUTES exporting?

also need help exporting only top (crank) channel

[image]

rusefillc, 2024-04-12:
> which I guarantee can export as csv as well

Does this warranty apply to "export only edges" @mck1117 ?

rusefillc, 2024-04-18:
We have .csv and automated test

closed 2024-04-18 by rusefillc

rusefillc, 2024-04-23:
> his screenshots are of sigrok, which I guarantee can export as csv as well

https://github.com/rusefi/rusefi/issues/6403