rusefi/rusefi #6360: Using LuaGauge1 as input to a Boost Blend Table specifically

ElDominio, 2024-04-13:
**Describe the bug**
Using LuaGauge1 as blend parameter is not working, only with Boost blend tables, seems fine with ignition and fuel. This happens with boost control active or inactive

**Hardware**
 - Board: [H154]
 - Engine info [Hyundai Theta]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Advanced
2. Boost blend tables
3. Set blend parameter to Luaauge1
4. See how any change in Luaauge1 is not reflected

**Expected behavior**
Be able to use LuaGauge1 as a blend parameter to add boost in different values of LuaGauge

**Screenshots**
[image]
Here we can see  x axis not tracking with LuaGauge1

[image]

Here we can see Fuel working correctly, but boost isn't

**Additional context**
Tune at https://rusefi.com/online/view.php?msq=1640

rusefillc, 2024-04-13:
@ElDominio please link a log for all that

ElDominio, 2024-04-13:
What does the log needs to show, LuaGauge1 changing ?

rusefillc, 2024-04-13:
> What does the log needs to show, LuaGauge1 changing ?

I am looking to see boostOpenLoopBlendParameter1 in the log

ElDominio, 2024-04-19:
https://rusefi.com/online/view.php?log=1378 log with luagauge1 changing

ElDominio, 2024-04-26:
was a me problem, i mixed up "param" and " bias", still a bit cryptic