rusefi/rusefi #6371: PCF8575 vs PCA9555APW

rusefillc, 2024-04-19:
we have
```
	// looks like we support PCF8575 i2c I/O expander
	m_i2c.read(/*address*/0x20, variant, sizeof(variant));
```

1) what is magic 0x20 where is it mentioned on https://www.ti.com/lit/ds/symlink/pcf8575.pdf?ts=1713412507841?
2) what about JLC C2652295 PCA9555APW is it compatible or how do we make it work?

VFRenjoyer, 2024-04-19:
[image]
Page 16 on the datasheet has this table defining the address. The default address is 0100000, and the last three bits can be set with pins A2,A1,A0

rusefillc, 2024-04-19:
0100xxx0?

rusefillc, 2024-04-19:
Ohh

[image]

See also https://forum.arduino.cc/t/pcf8574p-or-pca9555-creating-additional-do/89128/5

dron0gus, 2024-07-29:
Done on https://github.com/rusefi/rusefi/pull/6714

closed 2024-07-29 by rusefillc