rusefi/rusefi #6376: hellen on a bike: SD card logs report -32v on secondary TPS and secondary PPS

rusefillc, 2024-04-19:
WAT?!

[image]

rusefillc, 2024-04-19:
[re_13.zip](https://github.com/rusefi/rusefi/files/15046274/re_13.zip)

rusefillc, 2024-04-19:
decoded TPS/PPS and error counter follow the broken voltage so it's more than just a logging glitch

rusefillc, 2024-04-21:
Related to https://github.com/rusefi/rusefi/issues/6394

rusefillc, 2024-04-21:
[image]

rusefillc, 2024-04-21:
hello Bi Quad filter? funny how we have seem similar (?) negative recently https://github.com/rusefi/rusefi/issues/4117#issuecomment-2050459260

rusefillc, 2024-04-22:
rawPpsSecondary is with Bi Quad filter applied negative value tells us that at least one very noisy data point was received

200Hz sounds about right for this application

new data points in the logs: two rawRawPps fields which are completely unprocessed ADC values. Now we need a lucky log with those showing a problem

rusefillc, 2024-04-22:
currently ``#define ADC_SAMPLING_FAST ADC_SAMPLE_28``

```
#define ADC_SAMPLE_3            0   /**< @brief 3 cycles sampling time.     */
#define ADC_SAMPLE_15           1   /**< @brief 15 cycles sampling time.    */
#define ADC_SAMPLE_28           2   /**< @brief 28 cycles sampling time.    */
#define ADC_SAMPLE_56           3   /**< @brief 56 cycles sampling time.    */
#define ADC_SAMPLE_84           4   /**< @brief 84 cycles sampling time.    */
#define ADC_SAMPLE_112          5   /**< @brief 112 cycles sampling time.   */
#define ADC_SAMPLE_144          6   /**< @brief 144 cycles sampling time.   */
#define ADC_SAMPLE_480          7   /**< @brief 480 cycles sampling time.   */
```

rusefillc, 2024-04-22:
@dron0gus 

[image]

rusefillc, 2024-04-22:
[image]

andreika-git, 2024-04-22:
Maybe we should somehow check fastAdc.errorsCount which gets incremented on ADC conversion errors?

andreika-git, 2024-04-22:
One of possible reasons - ADC buffer memory corruption?..

rusefillc, 2024-04-22:
[image]

dron0gus, 2024-04-22:
discovery f407. tps1 and tps2 set to PA0..PA3.
Just press enable self-stimulation, disable few times. Get +1 fastAdcErrorCounter.
Fast ADC is not in correct state when timer callback is executed. ADC is in ADC_ACTIVE state.
[image]

dron0gus, 2024-04-22:
https://github.com/rusefi/rusefi/blob/294040c22a407e389a6f10faea0d052d163a782a/firmware/hw_layer/adc/adc_inputs.cpp#L176 why this was commented out? Cause to much noise?

rusefillc, 2024-04-23:
> why this was commented out? Cause to much noise?

Just created separate issue for that https://github.com/rusefi/rusefi/issues/6399 just in case

rusefillc, 2024-07-10:
[image]

rusefillc, 2024-07-10:
@dron0gus update: maaaaaaaaybe only happens while SD card logging

rusefillc, 2024-07-11:
[image]

rusefillc, 2024-07-11:
[image]

[image]

ElDominio, 2024-07-13:
> @dron0gus update: maaaaaaaaybe only happens while SD card logging

Just in case i barely ever use sd logging

dron0gus, 2024-07-13:
Looks like really related to SD/MMC access.
[image]

dron0gus, 2024-07-13:
Possible reason https://github.com/rusefi/rusefi/pull/6712

commit aaac8148 (2024-07-14): ADC array out of bounds when using analog muxes fix #6376
closed 2024-07-14 by rusefillc