rusefi/rusefi #6402: Ford Coyote Exhaust VVT position error

ElDominio, 2024-04-23:
**Describe the bug**
Exhaust VVT position is resolved correctly up to 20 degrees; after that, there is an error and VVT angle jumps about.

**Hardware**
 - Board: 8chan revC
 - Engine info Ford Coyote

**To Reproduce**
Steps to reproduce the behavior:
1. Set VVT trigger type to Ford Coyote
2. Enable exhaust VR cams
3. rev engine and move VVT position using actuators

**Expected behavior**
VVT angle is linear in the retard direction

**Additional context**
Tune and log at:
https://rusefi.com/online/view.php?log=1379

rusefillc, 2024-04-25:
[image]

rusefillc, 2024-04-25:
We see CAM sync errors

```
	CUSTOM_CAM_TOO_MANY_TEETH = 9004,
	CUSTOM_CAM_NOT_ENOUGH_TEETH = 9005,
```

let's have only one VVT input and output active and calibrate it to be vvt bank 1 intake even while in could be an exhaust, and grab a new log

rusefillc, 2024-04-25:
Looks like you are waiting for https://github.com/rusefi/rusefi/issues/5333

ElDominio, 2024-04-25:
Fix about #5333 worked, intake cams are working as expected; exhaust still crazy.

Log of ONLY bank 1 exhaust as bank 1 intake at 
https://rusefi.com/online/view.php?log=1380

rusefillc, 2024-04-28:
vvt resync happens meanings false vvt sync

[image]

ElDominio, 2024-05-01:
[all waves coyote ch1 crank ch2 intake cam ch3 exh cam.zip](https://github.com/rusefi/rusefi/files/15173760/all.waves.coyote.ch1.crank.ch2.intake.cam.ch3.exh.cam.zip)

Channel 1 is 36-1 VR 9924
Channel 2 is Coyote Intake Cam
Channel 3 is VR2, going through discrete

rusefillc, 2024-05-01:
@ElDominio note the differences in falls

[image]

ElDominio, 2024-05-01:
yes.......................................................................................?

ElDominio, 2024-05-31:
Bumping this because i want to dyno the car again

ElDominio, 2024-07-06:
This issue was caused because VR threshold was incorrecetly set up

[image]

With these settings, exhaust VVT reads fine , tested up to -35

closed 2024-07-06 by rusefillc