rusefi/rusefi #6409: Fix scheduling_s::momentX field type from efitick_t to efitimeus_t

kifir23917, 2024-04-24:
It looks like in live code we initialize [scheduling_s::momentX](https://github.com/kifir23917/rusefi/blob/43c0bda0027cc33b137ae481d9d321da3faa67fd/firmware/controllers/system/timer/scheduler.h#L69-L72) field with microseconds, not with ticks: see [here](https://github.com/kifir23917/rusefi/blob/2b80929ac6676577157f934ea65809535dccba57/firmware/controllers/system/timer/event_queue.cpp#L95).
There exists [code](https://github.com/kifir23917/rusefi/blob/bc2a2b9ab9b9e0b0b2d33c9d0cfa303c1d989cb2/firmware/controllers/system/timer/single_timer_executor.cpp#L85)  where we ititialize this field with ticks, but it is under `EFI_SIGNAL_EXECUTOR_ONE_TIMER` compilation flag - is it still actual?

mck1117, 2024-04-25:
> [here](https://github.com/kifir23917/rusefi/blob/2b80929ac6676577157f934ea65809535dccba57/firmware/controllers/system/timer/event_queue.cpp#L95)

those are ticks, no?

rusefillc, 2024-04-25:
> 
> those are ticks, no?

it depends

```
schedulingQueue.insertTask(scheduling, timeUs, action);
```

rusefillc, 2024-04-25:
PS: I sense a template incoming?

mck1117, 2024-04-25:
> PS: I sense a template incoming?

overkill, tick and us are both int64_t

rusefillc, 2024-04-25:
> overkill, tick and us are both int64_t

a template with a struct sprinkled to get strict typing?

rusefillc, 2024-06-14:
resolved

closed 2024-06-14 by rusefillc