rusefi/rusefi #6412: SD card logging is not reliable

rusefillc, 2024-04-24:
75% of the time we cannot access SD card logs via MSD

unclear is that's a Windows issue or us not closing file properly issue

more info needed

mck1117, 2024-04-25:
> we cannot access SD card logs via MSD

can you clarify what this means?

can't mount card? can't open file? open file but contents corrupt? open file but contents unexpected?

rusefillc, 2024-04-25:
> can't mount card? can't open file? open file but contents corrupt? open file but contents unexpected?

Windows 11 suggest formatting SD card instead of listing files

benassi124, 2024-04-25:
[image][image]

Sorry for potato pics

ggurov, 2024-04-25:
what happens when you pull the card out and put it into a sd card adapter ? card needs to be healthy for this to work right

benassi124, 2024-04-25:
> what happens when you pull the card out and put it into a sd card adapter ? card needs to be healthy for this to work right

Card works. Was in a 3d printer prior also tried different card

zed65, 2024-04-25:
I've never had any problems with that, tested with different SD-card and a xtsd01g. Though I'm on discovery-F407 firmware