rusefi/rusefi #6423: Launch control fuel addition does not work

ElDominio, 2024-04-26:
**Describe the bug**
Fuel is not being added even though I requested 100% extra fuel on launch control

**Hardware**
 - Board: 4chan 
 - Engine info 13B

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Advanced -> Launch Control
2. Enable it
3. add some fuel in "fuel added"
4. rev to launch control area
5. see no fuel is increased

**Expected behavior**
There is additional fuel added, base off current calculated fuel (take current fuel demand and add/remove whatever % is in "fuel added"

**Screenshots**
[image]

**Additional context**
I think this is related to not having the "fuel cut" option set to true. This feels like the problem where if ignition cut was not set true, there would not be timing retard.

zed65, 2024-04-26:
On my board with discovery firmware it works as expected (adds 100% fuel).

ElDominio, 2024-04-27:
What are the settings for your launch control?

zed65, 2024-04-27:
[image]
I tried both with and without fuel and ign cut, all worked.

ElDominio, 2024-04-27:
tf, well I attached the log, and the pulse width basically changes tracking the VE table

zed65, 2024-04-27:
I tried to load your tune on my board, launch worked.
Due to my stimulator I have to set my launch rpm to 2500
[launchTest.zip](https://github.com/rusefi/rusefi/files/15138073/launchTest.zip)

ElDominio, 2024-06-14:
As far as I understand this was my mistake?

closed 2024-06-14 by ElDominio