rusefi/rusefi #6424: CLT/IAT boost duty/target compensation

ElDominio, 2024-04-26:
**Is your feature request related to a problem? Please describe.**
Boost duty/target should have a 5x1 table to compensate for changes in temperatures (colder weather means air is more dense = same amount of DC get you more air; and vs CLT some people like to not have full power until operating temps are reached)

**Would we need new configuration fields?**
Two new tables to alter the duty (if in open loop boost mode) or target (if closed loop boost mode) to allow for higher/lower targets depending on temperature

**Describe the solution you'd like**
The said table to be added

**Describe alternatives you've considered**
I am currently using a blend table for this

kifir23917, 2024-07-22:
@ElDominio, am I correct that we need to add couple of tables similar to the following, but these two tables should contain 5 values instead of 16?
[image]
[image]

What should be names of these new tables that will be displaed on the top like `Intake air temperature fuel Multiplier` and `Warmup fuel manual Multiplier` on the tables above?

What should be default values in `Air Temp` and `Coolant` columns?

As I understand, in the one new table `Multiplier` column should be renamed to `Duty`, in the another new table `Multiplier` column should be renamed to `Target`, but I'm not sure which of these columns should be alongside with `Air Temp` and which of these columns should be alongside with `Coolant`. Could you please clarify it?

ElDominio, 2024-07-23:
Intake Air Temperature Boost Multiplier
Coolant Temperature Boost Multiplier

Second columns of should BOTH still say Multiplier, since they are multipliers to the base duty and changing the name to duty will only result in confusion.
Both of these should be defaulted to 1.00, and they should work only with Boost Duty, and not touch target.

Default values should be
20 - 1.00
40 - 1.00
60 - 1.00
80 - 1.00
100 - 1.00

rusefillc, 2024-07-26:
[image]

rusefillc, 2024-07-26:
@ElDominio in "open plus closed" which should we multiply? both open duty and closed target by same multiplier?

kifir23917, 2024-07-29:
Current implementation uses the following formulas:
### closed loop target
(`BOOST_CONTROL_TARGET_COMPONENT` + `BLENDS_COMPONENT`) * `LUA_BOOST_TARGET_MULT` + `LUA_BOOST_TARGET_ADD`
where:
- `BOOST_CONTROL_TARGET_COMPONENT` is calculated from `Boost control target (kPa)` table
[image]
- `BLENDS_COMPONENT` is calculated from blend tables
- `LUA_BOOST_TARGET_MULT` is configured with `setBoostTargetMult` Lua hook
- `LUA_BOOST_TARGET_ADD` is configured with `setBoostTargetAdd` Lua hook
### open loop duty
`LUA_BOOST_DUTY_ADD` + `BOOST_CONTROL_DUTY_CYCLE_COMPONENT` + `BLENDS_COMPONENT` + `GEAR_ADDER_COMPONENT`
where:
- `LUA_BOOST_DUTY_ADD`  is configured with `setBoostDutyAdd` Lua hook
- `BOOST_CONTROL_DUTY_CYCLE_COMPONENT` is calculated from `Boost control duty cycle (open loop)` table
[image]
- `BLENDS_COMPONENT` is calculated from blend tables
- `GEAR_ADDER_COMPONENT` is calculated from `Boost control open loop gear adder`
[image]

In the scope of this issue we're adding two new components:
- `CLT_BOOST_MULTIPLIER` is calculated from `Coolant Temperature Boost Multiplier` table
[image]
- `IAT_BOOST_MULTIPLIER` is calculated from `Intake Air Temperature Boost Multiplier` table
[image]

Am I correct that new formulas should be the following?
### closed loop target
(`BOOST_CONTROL_TARGET_COMPONENT` * `CLT_BOOST_MULTIPLIER` * `IAT_BOOST_MULTIPLIER` + `BLENDS_COMPONENT`) * `LUA_BOOST_TARGET_MULT` + `LUA_BOOST_TARGET_ADD`
### open loop duty
`LUA_BOOST_DUTY_ADD` + (`BOOST_CONTROL_DUTY_CYCLE_COMPONENT` * `CLT_BOOST_MULTIPLIER` * `IAT_BOOST_MULTIPLIER`) + `BLENDS_COMPONENT` + `GEAR_ADDER_COMPONENT`

ElDominio, 2024-07-29:
While the formulas look correct; Closed Loop TARGET should NOT be changed with the same tables as DUTY. They have to change independently (duty/target) in relation with CLT/IAT. This means we need one set of tables, another for duty. And anything that changes TARGET needs to be additive 

So
Duty = changes with multiplicative ratio, as shown in `LUA_BOOST_DUTY_ADD + (BOOST_CONTROL_DUTY_CYCLE_COMPONENT * CLT_BOOST_MULTIPLIER * IAT_BOOST_MULTIPLIER) + BLENDS_COMPONENT + GEAR_ADDER_COMPONENT`

Target = changes with additive, meaning `(BOOST_CONTROL_TARGET_COMPONENT  + BLENDS_COMPONENT) * LUA_BOOST_TARGET_MULT + LUA_BOOST_TARGET_ADD + CLT_BOOST_ADD + IAT_BOOST_ADD`

kifir23917, 2024-07-29:
So actually we need four new tables. And by the moment we have titles only for two tables:
- `Intake Air Temperature Boost Multiplier`
- `Coolant Temperature Boost Multiplier`
What should be labels for another two tables? Something like the following?
- `Intake Air Temperature Boost Adder`
- `Coolant Temperature Boost Adder`

Do we need to clarify in the titles which tables are for open loop and which tables are for closed loop?

rusefillc, 2024-07-29:
@ElDominio how confident are you that at least one human who is not you would use these four new curves? I wonder if our energy would be better spend working on custom curve usability and Lua integration?

ElDominio, 2024-07-29:
The thing is, if we are tackling OEM-replacement ECUs, this would be pretty useful. I think having this allows a higher quality tier of tuner to produce reliable tunes

kifir23917, 2024-08-02:
@ElDominio, I've supported the following curves in the scope of this issue:
### Open Loop
[image]
[image]
### Closed Loop
[image]
[image]

Is the naming of curves above good for you?

ElDominio, 2024-08-03:
I think the open loop ones should say "Boost Duty Multiplier" and the closed loop ones should say "Boost Target Adder" for clarity's sake

kifir23917, 2024-08-14:
@ElDominio, could you please check that all is OK with this functionality?

ElDominio, 2024-08-17:
Can you make the target adders only enabled if mode Open + Closed Loop is active?

enabling closed loop causes this
[image]

multipliers do work, but closed loop is now broken; this only shows up when KEYED ON! if you try this with key off it will not throw the error

[testadders.zip](https://github.com/user-attachments/files/16646338/testadders.zip)

kifir23917, 2024-08-19:
@ElDominio , I'm working on fixing `Access to uninitialized bins: clt (adder)` error. In case if i fix it, are you still sure that we really need to make the target adders enabled only if mode Open + Closed Loop is active?

ElDominio, 2024-08-19:
> @ElDominio , I'm working on fixing `Access to uninitialized bins: clt (adder)` error. In case if i fix it, are you still sure that we really need to make the target adders enabled only if mode Open + Closed Loop is active?

Yes, since in Open Loop there are no targets to be modified

kifir23917, 2024-08-20:
@ElDominio:
1. I hope that `Access to uninitialized bins: clt (adder)` error is fixed
2. Now target adders are only enabled only if mode `Open + Closed` Loop is active

Could you please test latest build?

ElDominio, 2024-08-22:
Confirmed all working fine!

Please rename the multipliers so they say "boost duty duty multiplier" and and "boost target adder"

kifir23917, 2024-08-23:
> Please rename the multipliers so they say "boost duty duty multiplier" and and "boost target adder"
1. I believe that you mean `boost duty multiplier`, not `boost duty duty multiplier`
2. Is my understanding correct that you want to rename column, marked on the screenshot below?
[image]

ElDominio, 2024-08-23:
Only the menu dropdown, not the column

kifir23917, 2024-08-26:
> Please rename the multipliers so they say "boost duty duty multiplier" and and "boost target adder"
> Only the menu dropdown, not the column

@ElDominio, I've renamed menu items:

[image]

closed 2024-08-26 by kifir23917