rusefi/rusefi #6451: Oscilations in DWELL length after 40+ hours of uptime

ggurov, 2024-05-02:
**Describe the bug**
It appears after some running, on the bench, dwell shows oscilations in duration consistent with a pattern:

commanded dwell: 3.5

3.5ms -> 3.2ms -> 3.0ms -> 2.7ms -> 2.4ms -> 2.1ms  -> 3.5ms

repeating on all channels. 

**Hardware**
 - Board: 407disco 
 - Engine info: 2jz-gte 6 cyl waste spark batch injection

**To Reproduce**
Steps to reproduce the behavior:
1. Tun board for 40+ hours on stim
2. Observe dwell duration on logic analyzer
3. `andrey_rusefi: WHAT THE HELL DWELL IS GOING NUTS`

**Expected behavior**
Dwell to not go nuts and stay consistent with commanded dwell.

**Screenshots**

[image]

[image]

ch 3,4,5,6,7,8 are ignition channels, waste spark so 1+6, 2+5, 3+4 are paired 

saleae capture: 

[dwell_oscilations.zip](https://github.com/rusefi/rusefi/files/15189382/dwell_oscilations.zip)

ts log:

[2024-05-02_09.49.51.zip](https://github.com/rusefi/rusefi/files/15189374/2024-05-02_09.49.51.zip)

overdwell counter seems to be ticking up at the time

[image]

https://youtu.be/rxIXa3DA6bA video of this happening

rusefillc, 2024-05-02:
I guess https://github.com/rusefi/rusefi/issues/6450 is somewhat of a preparation to start looking into this issue

rusefillc, 2024-05-03:
@kifir23917 https://www.youtube.com/watch?v=rxIXa3DA6bA

rusefillc, 2024-05-03:
saleae "saleae logic 1" https://support.saleae.com/logic-software/legacy-software/older-software-releases#logic-1-x-download-links

@kifir23917 funny thing that only spark is broken that must be telling us something
* looks like signal generation is not broken
* interesting thing if bad dwell times are not broken?
* would a unit test with very quick time movement help us here?

rusefillc, 2024-05-03:
@ggurov can you attempt reproducing with "one tooth" trigger instead of real 36-smth?

ggurov, 2024-05-04:
[image]

running

ggurov, 2024-05-05:
reproduced, similar thing: 

[image]

[image]

https://www.youtube.com/watch?v=kIxebA81Nag

[reproduced_oscillations_3_0_trigger.zip](https://github.com/rusefi/rusefi/files/15213943/reproduced_oscillations_3_0_trigger.zip)

tune log and .sal file in the zip above. 

"time" variable in TS shows about 50 hours.

rusefillc, 2024-05-07:
@ggurov @kifir23917 
* was it external signal generator? now that we use simple trigger shape can we attempt reproducing with self-stimulator?
* wonder if we can reproduce with self-stimulation on rusEFI simulator

ggurov, 2024-05-07:
external generator off, internal one running

[image]

[image]

running, output signals match what it was with external generator

rusefillc, 2024-05-07:
[image]

ggurov, 2024-05-07:
if that's my log, that's TS chugging at the start, check out SECL variable, it will do the same thing. maybe time does it too. it's an artifact of starting to log.

rusefillc, 2024-05-07:
@ggurov please run with "debug mode" set to "dwell metric" (see bottom right corner of weird dialog)

(that's legacy extra logging)

[image]

ggurov, 2024-05-07:
hmm, no secL variable ? there's gotta be something that ticks up from ecu monotonically...

rusefillc, 2024-05-07:
@ggurov see uneven duration of second in bottom yellow "uptime" line, looks like a gap in the log?

[image]

ggurov, 2024-05-08:
reproduced with internal stim, will post artifacts in the morning

ggurov, 2024-05-08:
INTERESTING THING TO NOTE:

once in this state, settings internal stim to 893 rpm = NO oscillations
894 = oscillations.

overdwell does not tick up @ 893, and starts going up at 894 rpm

this should be in the logs.

[image]

https://www.youtube.com/watch?v=sJhYllzm8cI

[dwell_oscillations_internal_stim.zip](https://github.com/rusefi/rusefi/files/15247749/dwell_oscillations_internal_stim.zip)

ggurov, 2024-05-08:
[2024-05-08_06.58.38-external-stim.zip](https://github.com/rusefi/rusefi/files/15247881/2024-05-08_06.58.38-external-stim.zip)

enabling external stim in that stage still produces the same results.

one thing to note, i've switched back to 36-2 for external stim, and consequentally internal stim.  and wave pattern seems consistent with the trigger pattern.

rusefillc, 2024-05-08:
[image]
[reference-issue-6451.zip](https://github.com/rusefi/rusefi/files/15249468/reference-issue-6451.zip)

ggurov, 2024-05-08:
https://rusefi.com/online/view.php?msq=1648 that was the tune with internal stim. it's also curtune.msq in the zip posted above with 893 vs 894 rpm separation

rusefillc, 2024-05-08:
"good" orange from top screenshot

[image]
pink 100 to 230, duration 130
orange from 330 to 480, duration 150

offset end of pink to end of orange 250

rusefillc, 2024-05-08:
"bad" orange

[image]

pink 100 to 250 duration 150
orange 340 to 450 duration 110

offset start of pink to start of orange 240
offset end of pink to end of orange 200

ggurov, 2024-05-10:
reproduced with engint_type 93 on a bare discover board:

[image]

[dwell_oscillations_discovery_engine_93.zip](https://github.com/rusefi/rusefi/files/15273869/dwell_oscillations_discovery_engine_93.zip)

logic analyzer log, tune, and log in the zip

ch2 and ch5 seem to be affected.

runtime of about 36 hours so..

issue happens above 3700 rpm

switching to internal trigger shows the same effect once ECU enters this state.

ggurov, 2024-05-10:
[Session 1-increasing-partial-dwell.zip](https://github.com/rusefi/rusefi/files/15273961/Session.1-increasing-partial-dwell.zip)

above a certain RPM, (5500 or so)

there is an increasing dwell length of one of the pulses

[image]

rusefillc, 2024-05-10:
sadDwellRatioCounter is alive. we've made progress!

[image]

rusefillc, 2024-05-10:
@ggurov why was RPM zero in the beginning?

what's going on with MAP sensor? I wonder if causation or correlation with sadDwellRatioCounter 

[image]

rusefillc, 2024-05-10:
Looks like issue depends on RPM

36-2 older tests, 893 rpm = NO oscillations, 894 = oscillations. 

TT_NARROW_SINGLE_TOOTH newer tests, threshold is around 3700

rusefillc, 2024-05-10:
let's keep using engine 93 narrow tooth

let's try 12000RPM? 15K rpm?

I am hesitant but focus should probably be on internal stimulator

Looks like issue depends on RPM
36-2 older tests, 893 rpm = NO oscillations, 894 = oscillations. 
TT_NARROW_SINGLE_TOOTH newer tests, threshold is around 3700

rusefillc, 2024-05-10:
WOW maybe smoking gun/maybe useful info?! why do we have overDwellCounter?!

let's try without

[image]

mck1117, 2024-05-10:
this is almost certainly a floating point loss of precision bug

rusefillc, 2024-05-10:
> this is almost certainly a floating point loss of precision bug

I know but what are we accumulating?

rusefillc, 2024-05-11:
@ggurov new smoke: 3700 threshold is when coil duty cycle crosses 25%

[image]

rusefillc, 2024-05-11:
also CUSTOM_OUT_OF_ORDER_COIL = 9012

rusefillc, 2024-05-11:
I've reproduced the issue

[image]

[image]

ggurov, 2024-05-11:
this is 767 12 cylinder @ 29000 rpm for approximately 19 hours

open up salea, look for this pattern:

ch 2 and ch3 are 40 seconds on, 40 seconds off, is at 14650 RPM precisely that should match up to the MLG

even going 10 rpm up from that causes a much more cyclical pattern

ch1 is cyl1, ch2 cyl2, ch3  cyl3, ch4 cyl4

1-2-3-4-5-6-7-8-9-10-11-12 firing pattern 12cyl

[image]

[767_26krpm_itnernal_single_tooth-14650-big-gaps.zip](https://github.com/rusefi/rusefi/files/15283266/767_26krpm_itnernal_single_tooth-14650-big-gaps.zip)

ggurov, 2024-05-11:
same as above, but added a slower log rate shorter log sal mlg

[image]
how linear uptime is depends on something secondary from RPM input.

as if it has to finish overflowing something, and then pace changes.

[767_26krpm_itnernal_single_tooth-aditional-slower-log-rate.zip](https://github.com/rusefi/rusefi/files/15283298/767_26krpm_itnernal_single_tooth-aditional-slower-log-rate.zip)

rusefillc, 2024-05-11:
@ggurov see EDIT2 in ticket description

mck1117, 2024-05-11:
https://github.com/FOME-Tech/fome-fw/pull/424 may be of interest

rusefillc, 2024-05-11:
> [FOME-Tech/fome-fw#424](https://github.com/FOME-Tech/fome-fw/pull/424) may be of interest

appreciate it!

rusefillc, 2024-05-12:
@ggurov there is a chance that this issue is now resolved. Are you available to test?

commit 087ebef9 (2024-05-14): Ignition scheduling errors after 40+ hours of uptime fix #6451
closed 2024-05-14 by rusefillc