rusefi/rusefi #6452: gcc13

rusefillc, 2024-05-02:
Do we support gcc13?

Which version of 13 has what problems?

looks like 13.2.Rel1 fails at runtime see https://github.com/rusefi/rusefi/pull/5849?

FDSoftware, 2024-10-29:
GCC 13.2.0 compiles ok but crashes at runtime on uaefi, I don't have a blue LED, nor do I have a USB device.

dmesg after upload:
```
[181821.311949] usb 1-9: Product: STM32  BOOTLOADER
[181821.311952] usb 1-9: Manufacturer: STMicroelectronics
[181821.311956] usb 1-9: SerialNumber: 205239A03732
[181925.708617] usb 1-9: USB disconnect, device number 15
``` 

```
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-linux-gnu/13/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 13.2.0-4ubuntu3' --with-bugurl=file:///usr/share/doc/gcc-13/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-13 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/libexec --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --enable-cet --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-13-XYspKM/gcc-13-13.2.0/debian/tmp-nvptx/usr,amdgcn-amdhsa=/build/gcc-13-XYspKM/gcc-13-13.2.0/debian/tmp-gcn/usr --enable-offload-defaulted --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu --with-build-config=bootstrap-lto-lean --enable-link-serialization=2
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 13.2.0 (Ubuntu 13.2.0-4ubuntu3) 
```

rusefillc, 2024-10-29:
@FDSoftware have you noticed https://github.com/rusefi/rusefi/pull/5849 progress or coincidence?

gcc13 is low priority :( it's probably asking for much better CI setup

FDSoftware, 2024-12-18:
small update, today i fucked up my docker installation so i'm testing with what i have installed locally, GCC 14.2 seems to work fine for compiling and HW CI

`gcc versión 14.2.1 20240912 (Red Hat 14.2.1-3) (GCC) `

hwci log:
https://gist.github.com/FDSoftware/68ed5adb499ce4f5b1021ccfba68f693

rusefillc, 2024-12-18:
On the one hand with current HW CI it looks somewhat safe to migrate, on the other hand I am not aware of _any_ benefit while the risk is always above zero. No plans to switch compiler until May of 2025

FDSoftware, 2024-12-18:
yep totally agree