rusefi/rusefi #6459: that 3+0 trigger name is horrible

rusefillc, 2024-05-03:
@ElDominio should it be "3+0 symmetrical" or "3+0 unsynchronized" or what? Same story for 6+0.

ElDominio, 2024-05-03:
How about "Symmetrical 3 tooth"?

And, why can't this kind of setup be detected using "Custom toothed wheel"

If there are no missing teeth, and ignition is not "single coil", assume we NEED VVT position and do not run until synchronized. I think that would be much more useable?

rusefillc, 2024-05-03:
```
#define operation_mode_e_FOUR_STROKE_CAM_SENSOR 2
#define operation_mode_e_FOUR_STROKE_CRANK_SENSOR 1
#define operation_mode_e_FOUR_STROKE_SIX_TIMES_CRANK_SENSOR 7
#define operation_mode_e_FOUR_STROKE_SYMMETRICAL_CRANK_SENSOR 4
#define operation_mode_e_FOUR_STROKE_THREE_TIMES_CRANK_SENSOR 5
#define operation_mode_e_FOUR_STROKE_TWELVE_TIMES_CRANK_SENSOR 6
```

See moon landing.