rusefi/rusefi #6477: Get rid of GTEST assertion wrappers

kifir23917, 2024-05-05:
We have various wrappers for GTEST assertions in our codebase (like EngineTestHelper::assertRpm or assertEqualsM2).
When test fails it outputs reference to wrapper instead of the failed row in test.

It's not convenient so I would suggest to get read of assertion wrapper.

rusefillc, 2024-05-05:
No doubt about ``assertEquals*``!

rusefillc, 2024-12-18:
@FDSoftware maybe?

FDSoftware, 2025-01-03:
@kifir23917 
[image]
These also need to be removed, right?

rusefillc, 2025-01-03:
>  These also need to be removed, right?

probably not since just macro without loosing file location?

FDSoftware, 2025-01-03:
cool I think with my last PR this issue would be ready then :fireworks:

rusefillc, 2025-01-03:
@FDSoftware what's the status of this? Is this now done?

FDSoftware, 2025-01-03:
yup, we can add more issues for test maintenance if necessary.

rusefillc, 2025-01-03:
@FDSoftware do you mean like https://github.com/rusefi/rusefi/issues/3553?

closed 2025-01-03 by rusefillc