rusefi/rusefi #6490: K6A (NON VVTI Trigger)

EA11R, 2024-05-09:
I would like to have this trigger wheel added on RUSEFI UAEFI firmware

[image]
[image]

Total tooth = 10  consisting of 3 pairs of tooth 3, 1 missing - 3 , 1 missing , 2 , 2 
it does read RPM as 12 - 2 

orientation of trigger in image is the way it mounts at cly #1 TDC 

areas free off tooth angle are around 70 deg for the largest gap

image of engine sniffer was taken of a hand rotated distributor 
I think I'm able to crank the engine with it tomorrow if it fits the F6A

rusefillc, 2024-05-10:
[image]

rusefillc, 2024-05-10:
[image]

rusefillc, 2024-05-10:
@EA11R I've made some progress no idea if it's right

https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers#suzuki-k6a

EA11R, 2024-05-10:
Yes, that's correct thank you very much 

the trigger is working correctly

commit b7ab2be3 (2024-05-10): only: Suzuki K6A (NON VVTI Trigger) fix #6490
closed 2024-05-10 by rusefillc